pihomeserver/Kupiki-Hotspot-Script #1: Script sans l'AccessPoint ?

toxic0berliner, 2016-09-16:
Bonjour, 
J'aimerais savoir s'il vous est possible de rendre ce script un peu plus "modulaire" pour me permettre de le tordre à mes besoins qui sont très similaires.
Grosso modo, je cherche à inverser le rôle de eth0 et wlan0 car le WAN viendrait de mon wifi partagé, et le hot-spot serait diffusé par une routeur physique (meilleur signal qu'une clef USB wifi) en mode AP non sécurisé et branché sur le port ethernet du raspberry. 

D'avance merci de tout aide.

pihomeserver, 2016-09-17:
Bonjour
Est-ce que si je vous envoie un script vous pouvez le tester et m'aider à debugger ? Car je n'ai pas cette configuration chez moi

toxic0berliner, 2016-09-17:
Le routeur qui fera access point est commandé mais pas encore arrivé. Donc
je ferai les essais dans la semaine prochaine et je suis dispo pour
échanger oui ;)

Merci de votre réponse si rapide !

pihomeserver, 2016-09-17:
Ok je suis en train d'optimiser et nettoyer le script je vais ajouter en local ensuite la configuration "inversée" pour les tests

toxic0berliner, 2016-10-12:
désolé du délai depuis la dernière fois...

Bonne nouvelle : j'ai réussi à faire ce que je voulais :D

Moins bonne nouvelle pour toi, j'ai du abandonner ce script...
Les raisons : j'ai prix une raspbian, qui plante en lançant ce script (un soucis de https pas possible pour apt-get), et du coup ça fout la merde car certains trucs sont faits et pas d'autres... 

Donc c'est une remarque : le script s'arrête au moindre problème, or le fait que la BDD existe déjà devrait pouvoir être ignoré...

Je suis donc reparti en combinant ton tutoriel texte avec la nouvelle version des commandes que tu mets dans ce script.

J'ai donc zappé la partie création de hotspot, j'ai pinaillé pour installer dd-wrt sur mon routeur premier prix configuré en point d'accès...

J'ai encore quelques points à fignoler, parceque je ne comprends pas vraiment le truc des 3 interfaces réseau...

Du coup, je me creuse la tête actuellement pour bloquer tout le traffic enprovenance du hotspot (eth1 sur le raspberry) pour le limiter aux seuls ports 80 et 443. Si tu as des idées je suis preneur !

Merci encore pour ta réactivité et ta proposition de script "inversé" !

pihomeserver, 2016-10-18:
Tu peux tenter de passer par un proxy qui pourra t'aider à faire le filtrage. Etape suivante chez moi qui est l'installation de Squid pour aider à filtrer aussi les sites accédés.

closed 2016-10-18 by pihomeserver