pihomeserver/Kupiki-Hotspot-Script #2: problem de copier coller ?

oliviusdm, 2016-09-17:
Super utile même si ça marche pas direct chez moi ;)  j'ai essayé de regarder pourquoi ça plantait après l'installation de mysql il me semble que cela vient des lignes ou  il manque un espace après le p : mysql -u root -**p$MYSQL_PASSWORD** mais vu que j'ai une connexion à mourir j'ai pas eu le temps de re-tester. j'ai aussi vu un problème il manque un point 127.0.**0.1** echo "GRANT ALL ON radius.* to 'radius'@'127.0.01';" >> /tmp/grant.sql    

par contre je connais pas trop comment fonctionne le portail captif mais je ne vois pas de conf server dhcp ...

;)

Olivier

pihomeserver, 2016-09-18:
Pour l'espace non il n'en faut pas. Le mot de passe doit bien être collé. Si ca ne fonctionne pas, est-ce qu'il y a moyen d'avoir la fin du fichier de log ? Il y a peut être un message plus explicite

Pour l'IP du localhost c'est corrigé dans la nouvelle version qui fonctionne car j'ai tout retesté hier soir depuis une minibian toute fraiche 

Le DHCP est inclus dans CoovaChilli

oliviusdm, 2016-09-18:
Ok donc c'est autre chose, Je reteste ca alors et te fait un retour pour le
soucis de mysql pour comprendre d'ou vient  mon problème.

oliviusdm, 2016-09-18:
Idem 
** Creating freeradius database
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: YES)

Je check encore ce soir et te fait un retour.

pihomeserver, 2016-09-18:
Plusieurs questions :
- Quelle distribution ?
- Est-ce que MySQL est déjà installé ?
- Si oui quelle est la liste des utilisateurs dans la table  user (avec la commande SELECT user, host FROM mysql.user;) ?

J'ai retesté encore aujourd'hui le script avec une minibian fraichement installé et j'ai aucun soucis
Pour info mon script est lancé en root

oliviusdm, 2016-09-18:
Bon bah la nickel sur distrib clean à part le jumpto "nextstep"  ligne 94 ;)

Beau boulot ;)

pihomeserver, 2016-09-18:
Ok je n'ai pas envoyé la bonne version. Merci de l'aide ;-)

closed 2016-09-18 by pihomeserver