pihomeserver/Kupiki-Hotspot-Script #3: Compatibilité rpi 2

Mr-Kha, 2016-09-20:
Bonsoir,
est-ce que ce script fonctionne avec un raspberry 2 et une clé wifi ?

merci d'avance.

pihomeserver, 2016-09-20:
Bonsoir
Il va y avoir une adaptation à faire dans le fichier de configuration de hostapd pour que le driver de la clé puisse se charger. Pour le reste pas de changement
Au départ les instructions étaient faites pour un modele 2
A tester ...

Peh4, 2016-10-17:
Bonjour, serait il possible de savoir quelle modification doit être faite pour le rpi2 ?
Merci

--
PA

pihomeserver, 2016-10-17:
@Peh4 vous savez quel driver pour la clé USB ? 
Les lignes à modifier sont :
- Ligne 105 en ajoutant le package avec les bons drivers pour qu'ils soient installés
- Ligne 235 en mettant le nom du driver

S'il n'y a pas de package, il faudra les installer manuellement avant de lancer le script (mais la modification ligne 235 reste à faire)

Peh4, 2016-10-17:
@pihomeserver merci pour votre réponse.
Mon dongle wifi est un TP-Link TL-WN725N / je vais chercher quel est le bon driver
Mais il est reconnu par le rpi2 sans drivers supplémentaires pour une utilisation classique

pihomeserver, 2016-10-18:
Si l'interface est effectivement détectée avant le démarrage du script, seule la ligne 235 avec le driver est à mettre à jour avec la valeur suivante :
```
driver=rtl871xdrv
```
(avec l'aide de https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=54946)

Mr-Kha, 2016-10-18:
bonsoir,
Comment faire le lien entre le modèle de la clé (reconnue par le rpi) et le driver à noter sur la ligne 235 ?
Merci pour ce super boulot et pour son partage.

pihomeserver, 2016-10-18:
Moi je passe par Google avec la référence de la clé et hostapd 
Et ensuite il faut regarder comment configurer la clé seule sous raspbian

Peh4, 2016-10-19:
Bonjour, j'ai testé la modification sur un Raspbian fraichement installé, mais ça bloque

** Updating system
** Upgrading all packages
** Checking if wlan0 interface already exists
** Update interface configuration
** Activating the WAN interface
** Activating the LAN interface
** Installing debconf tools
** Adding MySql password
** Adding MySql password (confirmation)
** Installing MySql, freeradius, iptables and other dependencies
**
** Something went wrong with the last command. Please check the log file
**

**voici ce que me disent les logs**
Réception de : 8 http://mirrordirector.raspbian.org/raspbian/ jessie/main libnl1 armhf 1.1-8 [122 kB]
Réception de : 9 http://mirrordirector.raspbian.org/raspbian/ jessie/main libgettextpo0 armhf 0.19.3-2 [115 kB]
Réception de : 10 http://mirrordirector.raspbian.org/raspbian/ jessie/main libdbd-mysql-perl armhf 4.028-2+deb8u2 [113 kB]
Réception de : 11 http://mirrordirector.raspbian.org/raspbian/ jessie/main libdbi-perl armhf 1.631-3+b1 [807 kB]
Réception de : 12 http://mirrordirector.raspbian.org/raspbian/ jessie/main mysql-client-5.5 armhf 5.5.52-0+deb8u1 [1 451 kB]
Réception de : 13 http://mirrordirector.raspbian.org/raspbian/ jessie/main libpcap0.8 armhf 1.6.2-2 [121 kB]
E: Le pilote pour la méthode /usr/lib/apt/methods/https n'a pu être trouvé.
**
** Something went wrong with the last command. Please check the log file
**

pihomeserver, 2016-10-19:
OK merci je vais mettre à jour le script en forçant l'installation de ce paquet

Peh4, 2016-10-19:
Merci pour votre aide, c'est toujours en train de mouliner mais ça semble bien parti

pihomeserver, 2016-10-19:
C'est vraiment long pour tout installer. Un tail -f avec le fichier de log permet de suivre l'avancement

pihomeserver, 2016-10-26:
@Peh4 ok pour l'installation ?

Peh4, 2016-10-26:
L'installation se passe bien, mais je dois louper quelque chose car je n'ai pas de nouveau réseau wifi détecté. Je n'ai pas eu le temps d'investiguer plus.

pihomeserver, 2017-01-27:
@Peh4 ca fonctionne ?

Peh4, 2017-01-27:
non toujours pas, j'ai abandonné pour le moment mais j'y reviendrai probablement
On est d'accord que normalement, un nouveau réseau wifi devrait être détectable après l'installation ?

pihomeserver, 2017-01-27:
Oui avec le SSID défini dans le début de script

whytelove, 2017-01-28:
Bonjour, je me posais la question, le script est compatible avec le RPI 2 certes, mais qu'en est-il des version précédentes ? Je ne possède malheureusement qu'un RPI B+ ... 
Si j'installe le driver de ma clef avant ai-je une chance que le script fonctionne ?

pihomeserver, 2017-01-29:
@whytelove le plus simple est bien entendu d'essayer mais si le driver pour hostapd est correct je ne vois pas de problème car toute l'installation est standard

closed 2017-03-30 by pihomeserver