pihomeserver/Kupiki-Hotspot-Script #4: Exit: Illegal number: -1

polzy, 2016-09-28:
Hi, 
I have a problem with sur bash script  : 

** Activating the WAN interface
** Activating the LAN interface
** Installing debconf tools
**
** Something went wrong with the last command. Please check the log file
**
pihotspot.sh: 46: exit: Illegal number: -1

Can you halp me ? 
Thanks you

pihomeserver, 2016-09-28:
OK the exit code is not correct but something went wrong with the installation of debconf Can you please look at the last lines of the log file in /var/log/pihotspot.log ?
-------- Message d'origine --------De : polzy <notifications@github.com> Date : 28/09/2016  09:23  (GMT+01:00) À : pihomeserver/Pi-Hotspot <Pi-Hotspot@noreply.github.com> Objet : [pihomeserver/Pi-Hotspot] Exit: Illegal number: -1 (#4) 
Hi, 

I have a problem with sur bash script  : 

** Activating the WAN interface

** Activating the LAN interface

** Installing debconf tools

**

** Something went wrong with the last command. Please check the log file

**

pihotspot.sh: 46: exit: Illegal number: -1

Can you halp me ? 

Thanks you

pihomeserver, 2016-09-28:
OK the exit code is not correct but something went wrong with the installation of debconf Can you please look at the last lines of the log file in /var/log/pihotspot.log ?

polzy, 2016-09-29:
Hi, 
Yes, thanks you

** 
** Log file created : /var/log/pihotspot.log
** Use command 'tail -f /var/log/pihotspot.log' in a new console to get installation details
** Checking if wlan0 interface already exists
wlan0     Link encap:Ethernet  HWaddr 00:08:d3:4a:12:b1  
** Update interface configuration
** Activating the WAN interface
ifup: interface eth0 already configured
** Activating the LAN interface
RTNETLINK answers: File exists
Failed to bring up wlan0.
** Installing debconf tools
Reading package lists...
Building dependency tree...Segmentation fault
** 
** Something went wrong with the last command. Please check the log file
**

pihomeserver, 2016-09-29:
The problem comes from the installation of the tools debconf. apt-get crashed 
Try manually to install them, solve the issue (not related to the script) and try to run the script again
You can use the command 
`
apt-get install debconf-utils
`

pihomeserver, 2016-10-01:
The return code has been corrected. Thanks for the issue report

closed 2016-10-01 by pihomeserver