pihomeserver/Kupiki-Hotspot-Script #6: Syntax Error

Saulgodman, 2016-10-22:
I hab an error:

`./pihotspot.sh: line 265: syntax error near unexpected token `('
./pihotspot.sh: line 265: `display_message " (login : administrator / password : radius)"'
`

pihomeserver, 2016-10-22:
Hello
Not sure what script you are using at line 265 is not this command. Take the latest version and try again.

Saulgodman, 2016-10-22:
Hello,

Strange I use the latest version on Minibian.

:(

pihomeserver, 2016-10-22:
Look at the source on GitHub and you will see that the line number does not get the instruction with the error
By the way if you reach this display line, it means that the hotspot is up and fully configured

Saulgodman, 2016-10-22:
Mh. Here is a picture of a clean installation.

[https://owncloud.blungogo.de/index.php/s/a8xchAOXo4hTC4y](url)

pihomeserver, 2016-10-22:
Ok one " is missing in the new instruction for Pi 2
Can you please give a new try ?

Saulgodman, 2016-10-22:
Yes now it runs! :)

Thank you very much

closed 2016-10-26 by pihomeserver