pihomeserver/Kupiki-Hotspot-Script #7: Redirect after login.

bobdmagic, 2016-10-24:
When you log into the hotspot it directs you to a successful login page. Then after a set time it directs you to another page. I would like prefer that it directs you to the google search page when it leaves the successful login page.  Also, I changed the title in the title file but it stayed the same after reboot. Where else do I change the title?
Thanks in advance for your reply.

pihomeserver, 2016-10-24:
Not sure but maybe you have to rebuild the www folder after each change (you speack about title.tmpl ?)
I still have to work on these customizations. If you have advices they are welcome ;-)

pihomeserver, 2017-06-27:
In the latest version, the portal used is in html with a redirection on success that you can customize

closed 2017-06-27 by pihomeserver