pihomeserver/Kupiki-Hotspot-Script #8: Warning after Chilli service Start

lasagh, 2016-11-03:
Hello!

I'm having trouble executing the script between Coova and hostapd installation on my RP3 B Model.

The execution stops and the kernel show me that warning in loop:
chilli: packet size is too short (42 <= 50)

It's the second time that I format my SD and start a new minibian installation before execute the script, and unfortunately got the same issue.

Does anybody know why?

pihomeserver, 2016-11-03:
Hello
After a Google search I found a way to solve it. It's not related to the script but to Coova : https://github.com/coova/coova-chilli/issues/261
Please try and tell me if it solve the issue. I will try to include a manual patch in the script.
If you still have issues, please try the Kupiki-Hotspot distro (https://github.com/pihomeserver/Kupiki-Hotspot)

pihomeserver, 2016-11-16:
@lasagh : did you solved your issue ?

lasagh, 2016-11-16:
Hello! 

Thanks for asking. 
I did yesterday! But, executing a manual installation.
MySQL + Freeradius + HOSTAPD + Coovachilli 

TY!

closed 2016-11-16 by lasagh