pihomeserver/Kupiki-Hotspot-Script #9: Utiliser une connexion wifi (wlan1) à la place de eth0 pour l'internet.

mospat, 2016-11-12:
Bonjour,

Je viens d'utiliser vos scripts pour transformer ma Pi en HotSpot et cela fonctionne très bien (aussi bien sur une pi3 qu'une pi2 après quelques adaptations). Bravo !

Toutefois j'aurais voulu utiliser également une deuxième connexion wifi (wlan1) comme connexion vers l'internet à la place de eth0, et là je ne suis pas certain des manipulations à effectuer.

Auriez-vous une piste ?

Merci d'avance

pihomeserver, 2016-11-12:
Bonjour

Il faut remplacer eth0 par wlan1 au début du script
```
WAN_INTERFACE="eth0"
```
par
```
WAN_INTERFACE="wlan1"
```
Mais il faut s'assurer que cette clé est bien installée et configurée en wlan1 sinon ca va mettre le bazar. Et le script pour le moment ne s'en occupe pas. Vous pouvez dans un premier temps configurer la clé, changer le script puis l'exécuter. Je suis preneur du retour/avancement !

mospat, 2016-11-13:
Ok merci pour la rapidité! Petits soucis avec la Pi3 (moins avec la Pi2), mais je reviens avec mes conclusions rapidement

mospat, 2016-11-19:
Voilà, ça fonctionne parfaitement sur une pi3.
Après l'installation du minibian, j'ai installé les pilotes pour ma clé (rtl8188cus) et monté ma connexion wifi sur wlan0. (connexion vers mon routeur internet)
Le script est modifié de telle sorte que :

WAN_INTERFACE="wlan0"
LAN_INTERFACE="wlan1"
execute_command "ifconfig -a | grep wlan1" false "Checking if wlan1 interface already exists" (Ligne 102)

Mais je reste connecté en filiaire via eth0 durant l'installation (sinon les reconfigurations réseaux me font perdre l'interface wlan0).

C'est donc parfait 👍  Un grand merci !

closed 2016-11-19 by mospat