pihomeserver/Kupiki-Hotspot-Script #10: Connected clients not able to browse the internet

arvindacumos, 2016-11-18:
Hello,

Thanks for the script it worked like a charm! Just one problem. Clients connected are not able to browse the internet. I am using the default configuration from the script. Clients can see the portal login and able to login but browsing the internet is not working on the client.

Thanks in advance.

pihomeserver, 2016-11-18:
Hi. Can you please check that the Pi still has an internet connection and that the iptables rule is well placed at the end of the script ?
Also did you try from different clients ? phone, tablet and pc

arvindacumos, 2016-11-22:
Everything seems fine, Rpi do have internet connection. I tried on the PC and I notice that every time I open a URL or click a link on an open webpage, I always presented with the Coova login page although I had already logged in. This maybe the reason why mobile apps were unable to connect to the internet. Is there a way to save the login credentials on the client side so there's no need to login every time you open a URL or click a link on an opened page?

pihomeserver, 2016-11-22:
Ok it seems that authentication has an issue. Can you please confirm that you created a user from daloradius interface, that the user exists (you can check in the interface how many users have been created and also how many are logged in)
If you are really sure that the user exists, please look at freeradius log. Also you can check authentication from command line with following commands (Replace SQL root paswword with your own) :
```
echo "insert into radcheck (username, attribute, op, value) values ('usertest', 'Cleartext-Password', ':=', 'passwd');" | mysql -u root -praspbian radius
```
and then (Replace testing123 with your secret key) :
```
radtest usertest passwd localhost 0 testing123
```
You should receive a token"Accept-Accept"
You can also test the user your created within daloradius

arvindacumos, 2016-11-22:
this is the response when i issue a radtest. I never got a token "Accept-Accept"

Sending Access-Request of id 9 to 127.0.0.1 port 1812
        User-Name = "test123"
        User-Password = "pass123"
        NAS-IP-Address = 127.0.1.1
        NAS-Port = 0
        Message-Authenticator = 0x00000000000000000000000000000000
Sending Access-Request of id 9 to 127.0.0.1 port 1812
        User-Name = "test123"
        User-Password = "pass123"
        NAS-IP-Address = 127.0.1.1
        NAS-Port = 0
        Message-Authenticator = 0x00000000000000000000000000000000
Sending Access-Request of id 9 to 127.0.0.1 port 1812
        User-Name = "test123"
        User-Password = "pass123"
        NAS-IP-Address = 127.0.1.1
        NAS-Port = 0
        Message-Authenticator = 0x00000000000000000000000000000000
radclient: no response from server for ID 9 socket 3

pihomeserver, 2016-11-22:
Did you try to investigate ? You can see in the message that the server freeradius does not respond
Can you check that free radius is running ? If not try to start it and check again. If still not running did you reboot ?
If free radius is not running i guess that the user creation failed too

pihomeserver, 2017-02-27:
@arvindacumos any news about the issue ?

closed 2017-03-30 by pihomeserver