pihomeserver/Kupiki-Hotspot-Script #12: Self-Registration not working

hijackindie, 2016-12-18:
I'm not sure what I am missing but when I set HS_REG_MODE="self" I get the link to register and can fill out the infos but nothing happens. The form is shown again with the previous inputs (except for the 2nd entry of the password). How can I set this up so a guest has to register with their email before adding a user account? And can the account be added automatically?
Thanks in advance for your help.

pihomeserver, 2016-12-18:
Hi.  Seems to be an issue for coovachilli more than the script.
Also did you check for the log if there is any revelant message ?

hijackindie, 2016-12-19:
Hi - which log? I am not sure where coova-chilli writes to. I cannot find anything like a webserver log. Thx.

pihomeserver, 2016-12-20:
Seems that coova has no log, only the one for free radius and not sure it could be helpful for you

Did you get a look at http://iantearle.com/blog/coova-v1-fix-captive-portal-self-registration ?

hijackindie, 2016-12-20:
Brilliant! It was the last line that was incorrect. Now the self-registration worked - at least for using a localuser file. Still need to try it with Radius. Thanks very much!

pihomeserver, 2016-12-20:
Thanks to you ! I will have to update the script to include the correction

closed 2016-12-20 by pihomeserver