pihomeserver/Kupiki-Hotspot-Script #13: Possible syntax error?

chseluv, 2017-01-19:
Thanks for your script, it's just super useful to me.

I'm using a RaspberryPi 3 with Raspbian Jessie Lite (2016-11-25)

http://pastebin.com/e70hP0ha

not a real expert here, maybe it's me doing something wrong?

Thanks for your work and your time

pihomeserver, 2017-01-19:
Hi
Does that mean that the script stops after the try to add the mysql password ?
No strange chars set in the password ?

chseluv, 2017-01-19:
Yes, exactly.
I did not the change the mysql password that the script came with for testing purposes, thus it's still a plain "pihotspot" and the installation did not go past that point

pihomeserver, 2017-01-19:
Please try the following command :
debconf-set-selections <<< 'mysql-server mysql-server/root_password password pihotspot'
echo $?

Please run the echo line right after the debconf to display the return code of the call

chseluv, 2017-01-19:
By using the command you wrote exactly as it is the echo outputs 1, when using "sudo" I get 0 instead

pihomeserver, 2017-01-19:
Ok my fault : how do you start the script ? Using root or using sudo script ? 
Even if it's not the best way, running the script as root should not raise error ...

chseluv, 2017-01-19:
I used sudo, I will try running the script as root and will get back to you

chseluv, 2017-01-19:
Unfortunately the same error occurs, http://pastebin.com/qWp2ksg4

pihomeserver, 2017-01-19:
I will have to test again what's wrong. If you don't want to wait, just copy paste commands one by one to execute them ...

chseluv, 2017-01-19:
Sure, I will try that too. Thanks for your help

scottie, 2017-01-19:
I also get same issue

**Adding MySql password
./pihotspot.sh 1: eval: Syntax error: redirection unexpected

scottie, 2017-01-19:
This is where the issue happens:

LINE 139

execute_command "debconf-set-selections <<< 'mysql-server mysql-server/root_password password $MYSQL_PASSWORD'" true "Adding MySql password"

pihomeserver, 2017-01-20:
@scottie @chseluv can you please log in a term and run the following command :
echo $SHELL

pihomeserver, 2017-01-20:
Also you can try by replacing both debconf-set-selections command :
```
execute_command "debconf-set-selections <<< 'mysql-server mysql-server/root_password password $MYSQL_PASSWORD'" true "Adding MySql password"
execute_command "debconf-set-selections <<< 'mysql-server mysql-server/root_password_again password $MYSQL_PASSWORD'" true "Adding MySql password (confirmation)"
```
with
```
execute_command "echo 'mysql-server mysql-server/root_password password $MYSQL_PASSWORD' | debconf-set-selections" true "Adding MySql password"
execute_command "echo 'mysql-server mysql-server/root_password_again password $MYSQL_PASSWORD' | debconf-set-selections" true "Adding MySql password (confirmation)"
```

scottie, 2017-01-21:
This issue is fixed now,
It was due to starting the script using SH.
If one  starts with SH it ignores the shebang and dosnt use the bash that has >>>.
One must chmod +x script.sh then simply ./script.sh and all is fine.

This issue is fixed but new issue is posted.
fix: 
sudo chmod +x script.sh && sudo ./script.sh

closed 2017-01-21 by pihomeserver