pihomeserver/Kupiki-Hotspot-Script #17: Ubuntu Mate Xenial support - RPi3

chseluv, 2017-01-26:
I launched the script on Ubuntu-Mate on a RPi3 and it seems some of the packages in would install have been renamed so the procedure stops at various stages.

I tried to pin them down and, if that helps, I'll list my suggestion.

Packages that no longer work and what I supposed may have superseded them

- libnl1 --> libnl-3-200 or libnl-3-200-udeb (not found in normal repositories)
- libnl-dev --> libnl-3-dev
- php5-fpm --> php-fpm
- php5-gd --> php-gd
- php5-mysql --> php-mysql
- php5-xcache --> php-cache
- libpcrecpp0 --> libpcrecpp0v5
- nginx --> "nginx-core" and "nginx-common"

Apart from that, I found that the script got stuck at line 269. By resuming manually I got to the end of the script ending up with a WiFi hotspot that doesn't connect devices to the internet or to coova chilli, and these services hadn't been set to start at boot time. Also, localhost/daloradius gave me a 502 error so I don't think this solution really worked

pihomeserver, 2017-01-26:
Hello
The solution worked and works on raspbian as explained in the reader. 
Thanks for.your comment about packages names with Ubuntu that I never tried

chseluv, 2017-01-26:
For some reason Ubuntu Mate Xenial handles USB modems way better than Raspbian, I'll give a try to the latest edition too. Hoping my packages will work as intended

pihomeserver, 2017-01-26:
OK great thanks for helping. Take care of lines that store mysql password there is a bug in need to fix it s an opened issue

chseluv, 2017-01-26:
On the latest version of Raspbian Jessie Lite (2017-01-11) this script gets stuck after it checks whether CoovaChilli created the tun0 interface and outputs "Unable to find chilli interface tun0", as instructed to.
No MySQL errors up to that line.

I cloned the script through git

pihomeserver, 2017-01-26:
The solution is in a closed issue that I have to add in the script. The minibian version does not have the issue I have to look at it

closed 2017-02-27 by pihomeserver