pihomeserver/Kupiki-Hotspot-Script #18: NAS configuration

psellers89, 2017-01-26:
Hello,
First thanks for your project, it was exactly what I was looking for !

I have a problem with the NAS configuration. When I try to disconnect a user from Reports > Online Users, I have this error : 
radclient: no response from server for ID 33 socket 3

I configured the NAS as follows : 
NAS IP/Host  : 127.0.0.1
NAS Secret : testing123
NAS Type  : other
NAS Shortname : localhost
NAS Ports : 3799

What am I doing wrong ?

pihomeserver, 2017-01-27:
First for my curiosity, what is your hardware, OS and did you get any bugs with the script ?

Do you want to disconnect the user or delete it ?

psellers89, 2017-01-27:
I am using the Pi 3, 
Linux minibian 4.4.38-v7+
I didn't get any bug with the script, well done!
I can connect with this command : 
`radtest testing password localhost 0 testing123`

But I can't disconnect a user from this page:
http://192.168.20.136/daloradius/config-maint-disconnect-user.php

Is it working for you?

Thanks again for your project. It made me saved a lot of time!!

pihomeserver, 2017-03-01:
Ok found partly why : coa site is not activated by default for free radius
-> link the file coa from /etc/freeradius/sites-available in /etc/freeradius/site-enabled
And restart freeradius you should now have an answer

Just Coova does not seem to receive the request so seems to not be enough ....

pihomeserver, 2017-03-02:
Just add 
```
HS_COAPORT=3799
```
in /etc/chilli/defaults and restart the service

closed 2017-03-02 by pihomeserver