pihomeserver/Kupiki-Hotspot-Script #19: prepaid user setup

reigelgallarde, 2017-02-24:
hello, thank you for the script, it's working.

I have a question though, how to setup/create a user with 1 hour limit?

pihomeserver, 2017-02-27:
Hello

Please look at Daloradius functions. An exemple of configuration that seems to work : https://ubuntuforums.org/showthread.php?t=2070298&highlight=daloradius

You have to create counters that will consolidate the time and volume of a connected user (section FreeRadius SQL counter.conf settings needed)

Then you create profiles linked to the counters and apply to the user (section daloRADIUS 0.9-9 – QUCIK START)

Hope it helps

reigelgallarde, 2017-02-27:
ok, thanks... I will try this...

I hope you can add this also on the script... would be very useful for everyone...

pihomeserver, 2017-02-27:
Yes that's a good idea. At least I can add the counters creation.

reigelgallarde, 2017-02-28:
I was not able to make it work on my first try. Haha. Just not my thing.
I hope you can update the script and add this kind of functionality.

I will however try again until it works. 😄

pihomeserver, 2017-02-28:
Yesterday I updated configuration files. Hope that today I will be able to create attributes in daloradius and test them

reigelgallarde, 2017-02-28:
great! wish you good luck!

reigelgallarde, 2017-03-09:
hello, have you got any luck on this one?

anyway, I have tested your new script on a fresh install of minibian, and I think it got some error...
freeradius is not running...
I tried calling `service freeradius start` and it won't start... 
searching at Google I found and tried to call `freeraduis -X`
freeradius will still not run... something with ports being used chilli. 

I then remove coa like `rm /etc/freeradius/sites-enabled/coa`
do a reboot, then all starts working... not sure what's wrong.

pihomeserver, 2017-03-09:
Updated. The Coova CoA port was in conflict with the freeradius support that is not required.

The new script activates defaults SQL counters. Look here for implementation : https://wiki.freeradius.org/modules/Rlm_sqlcounter

Please tell me if it works

reigelgallarde, 2017-03-09:
yes, this one works... I tried it earlier... using mysql commands....

what's not working is daloradius... it seems daloradius does not have the correct sql command... I'm not sure...

pihomeserver, 2017-03-09:
Ok if both of us can investiguate we should be able to find a way.
I will put on side the Squid integration for now to come back on this point

pihomeserver, 2017-03-09:
Checked daloradius and the radcheck table content and the attribute is well created but not taken into account to disconnect the user
[image]

pihomeserver, 2017-03-10:
In progress :
1) Edit /etc/freeradius/sites-enabled/default and uncomment line 192 to authorize daily counter
2) Restart freeradius
3) In daloradius add one attribute using the daloradius dictionary

[image]

For now i got disconnected after 30 seconds but able to reconnect ...

PR #20 (2017-03-30): Update pihotspot.sh
add code to update freeradius configs. So as to daloradius and freeradius can work together.
fixes #19 issue.
commit 41f79b4e (2017-03-30): Update pihotspot.sh

add code to update freeradius configs. So as to daloradius and freeradius can work together.
fixes #19 issue.
closed 2017-03-30 by pihomeserver