pihomeserver/Kupiki-Hotspot-Script #21: Log Loppsi

NuM82, 2017-03-28:
Bonjour,
Après une installation sans soucis particulier sur un RPI2 avec une jessie-lite toute fraîche et la modification des fichiers `/etc/network/interfaces` (les conf du wifi par défaut de la jessie rentre en conflit avec celles du script) et `/etc/hostapd/hostapd.conf` (p'tit soucis de driver propre à ma config) pour que cela tourne correctement, je teste un peu ce nouvel hotspot.
Y a encore quelques modifs à faire dans le script concernant le report des fichiers log dans DaloRadius (pb de droits d'accès de certains fichiers ou fichiers manquants) mais ceux ne sont que quelques détails.
Je me demande s'il y a quelque chose concernant les logs type Loppsi dans cette compilation hotspot ou s'il y a un truc que je n'aurai pas vu ou si cela n'est pas tout simplement encore implanté (si ça doit l'être) ?
Pour les générer, la (bonne) piste serait l'activation et triturage des logs d'iptables ou l'installation de squid (qui m'oblige d'utiliser un fichier `proxy.pac` pour réussir à log les requêtes https) ?

Pour le reste, chapeau bas pour la réalisation du script, je serai bien incapable pour le moment d'en faire autant.
Cordialement,
NuM82

pihomeserver, 2017-03-30:
Salut,
Peux tu donner les modifications liées à jessie-lite ? Je peux essayer d'intégrer ca dans le script

Je suis pas certain de comprendre le besoin des logs : tu veux filtrer les accès, juste les tracer ?
Dans les logs de freeradius tu as déjà des traces de qui se connecte

NuM82, 2017-03-30:
'lut,
Concernant le fichier `/etc/network/interfaces`, j'ai dû commenter les lignes suivantes inscrites dans le fichier d'origine de la jessie lite (2017-03-02-raspbian-jessie-lite.img)
```
allow-hotplug wlan0
iface wlan0 inet manual
     wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

allow-hotplug wlan1
iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
pour que ton ajout dans le fichier soit pris en compte

Pour le fichier `/etc/hostapd/hostapd.conf`, les modifs sont propres à ma config et ma clé usb wifi.

Pour les logs, pas besoin de filtrer les accès ... juste de tracer pour être en conformité avec la loi.
je t'invite à consulter ces 2 liens : 
https://www.debian-fr.org/t/partage-wifi-et-log-des-url-visitees/13223
http://www.aecom.org/Vous-informer/Juridique-TIC/Communications-electroniques/Technologies/Conditions-d-acces-a-des-reseaux-sans-fils-hotspot-et-Wifi-publics

Je vais jeter un oeil sur les logs de freeradius et revient vers toi dès que possible.
@+

pihomeserver, 2017-03-30:
Ok. Ca me confirme qu'il faut vraiment que je mette Squid dans la boucle

Les logs de freeradius ne vont pas (à vérifier) donner l'URL accédée. Ce n'est pas son rôle, qui est d’autoriser ou non l'accès

NuM82, 2017-03-30:
Je ne suis pas sûr qu'il faille inclure squid dans le package. 

Pour les logs de squid (j'ai mis ça en place y a 2 ans sur squid 3.4.8), il suffit d'ajouter `logformat loppsi %{%d/%m/%Y %H:%M:%S}tl  %>eui  %ru` dans le fichier `/etc/squid3/squid.conf` pour obtenir un fichier `var/log/loppsi/loppsi.log` contenant par exemple ceci : 
```
26/03/2017 16:23:08  54:27:1e:1f:ff:aa  http://tap2-cdn.rubiconproject.com/partner/scripts/rubicon/emily.html?
26/03/2017 16:23:08  54:27:1e:1f:ff:aa  cookiex.ngd.yahoo.com:443
```

Si on configure squid avec `http_port 192.168.10.1:3128 transparent`, on va pouvoir loguer seulement les requêtes http (du moins, je ne suis pas de mémoire parvenu à loguer les reqêtes https). Coté positif, aucun paramétrage des postes clients est nécessaire.

Si on configure squid avec `http_port 192.168.10.1:3128`, on va pouvoir loguer les requêtes http et https mais on sera obligé de configurer les postes clients avec un fichier `http://192.168.10.1/proxy.pac` ou `http://192.168.10.1/wpad.dat` contenant : 
```
function FindProxyForURL(url, host)
{
   if (isInNet(host, "192.168.10.0", "255.255.0.0")) {
      return "DIRECT";
   } else {
      if (shExpMatch(url, "http:*"))
         return "PROXY 192.168.10.1:3128" ;
      if (shExpMatch(url, "https:*"))
         return "PROXY 192.168.10.1:3128" ;
      if (shExpMatch(url, "ftp:*"))
         return "PROXY 192.168.10.1:3128" ;
      return "DIRECT";
   }
}```

Dans les 2 cas, on ne logue pas ce qui se passe ailleurs (mail, etc...).
C'est pour cela que je pense que la solution est du coté de l'activation des logs d'iptables (+ logs dhcpd) qu'il faudra triturer pour sortir quelque chose de correct.
@+

NuM82, 2017-04-01:
Pour générer les logs d'iptables, il a fallu que je modifie à la fois le fichier `/etc/chilli/up.sh `qui gère les règles iptables du hotspot (Attention à bien mettre la règle qui renvoie vers le log avant la règle originale) et le fichier `/etc/rsyslog.conf` pour éviter de pourrir les log system.
Voilà le résultat : 
```
Mar 31 19:31:58 raspberrypi kernel: [17041.037875] HTTP: IN=wlan0 OUT= MAC=00:c1:41:37:1a:ae:00:25:d3:d6:1a:af:08:00 SRC=192.168.10.3 DST=107.151.3.106 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26251 DF PROTO=TCP SPT=52675 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:58 raspberrypi kernel: [17041.037937] HTTP: IN=tun0 OUT= MAC= SRC=192.168.10.3 DST=107.151.3.106 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26251 DF PROTO=TCP SPT=52675 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:58 raspberrypi kernel: [17041.357821] HTTPS: IN=wlan0 OUT= MAC=00:c1:41:37:1a:ae:00:25:d3:d6:1a:af:08:00 SRC=192.168.10.3 DST=172.217.18.142 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26262 DF PROTO=TCP SPT=52677 DPT=443 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:58 raspberrypi kernel: [17041.357858] HTTPS: IN=tun0 OUT= MAC= SRC=192.168.10.3 DST=172.217.18.142 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26262 DF PROTO=TCP SPT=52677 DPT=443 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:58 raspberrypi kernel: [17041.374551] HTTP: IN=wlan0 OUT= MAC=00:c1:41:37:1a:ae:00:25:d3:d6:1a:af:08:00 SRC=192.168.10.3 DST=151.101.194.49 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26263 DF PROTO=TCP SPT=52678 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:58 raspberrypi kernel: [17041.374643] HTTP: IN=tun0 OUT= MAC= SRC=192.168.10.3 DST=151.101.194.49 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26263 DF PROTO=TCP SPT=52678 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:59 raspberrypi kernel: [17042.126780] HTTP: IN=wlan0 OUT= MAC=00:c1:41:37:1a:ae:00:25:d3:d6:1a:af:08:00 SRC=192.168.10.3 DST=151.101.66.49 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26336 DF PROTO=TCP SPT=52679 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
Mar 31 19:31:59 raspberrypi kernel: [17042.126808] HTTP: IN=tun0 OUT= MAC= SRC=192.168.10.3 DST=151.101.66.49 LEN=52 TOS=0x00 PREC=0x00 TTL=128 ID=26336 DF PROTO=TCP SPT=52679 DPT=80 WINDOW=8192 RES=0x00 SYN URGP=0 
```
Le problème est que pour une même page d'un site, on se retrouve avec une multitude de lignes dans le fichier de log et donc rapidement avec un gros fichier ...
Un autre problème est que l'IP ne correspond pas à une url ...
Bref, je ne crois pas qu'il s'agisse de la solution.
Je vais regarder du coté de tcpdump, de wireshark ou de Dsniff (Urlsnarf) pour voir si cela correspond mieux au besoin. Si quelqu'un à d'autres pistes, je suis preneur...

pihomeserver, 2018-02-04:
Logs are now included in the 1.5 release of the script

closed 2018-02-04 by pihomeserver

pihomeserver, 2019-01-06:
@petitours : look at issue #127 you will get information about the access.

For the retention period it's the configuration it's defined in the script parameter _NETFLOW_LOGS_DAYS_ (default is 1 year aka 365d). If you want to change it, please look at the _nfexpire_ command. That can update the value.