pihomeserver/Kupiki-Hotspot-Script #23: Use counters based on time

pihomeserver, 2017-04-01:
@reigelgallarde : can you please tell us how you used the counters for time basis ? On my side i tried with a counter called Max-All-Session set to 180 in the check attributes but it does not work and the user is not disconnected

reigelgallarde, 2017-06-05:
sorry for late reply but I followed this:

**Create Profiles – Time Based Profile**
Go to Management tab > Select Profiles > Create New Profiles >Add Profile Attributes
Type Profile Name, e.g. 60Mins

**Add attributes**
_Check Attributes_
Simultaneous-Use = 1
Max-All-Session = 3600
[this is in seconds, for 60mins = 3600seconds]
Session-Timeout = 3600
_Reply Attributes_
Session-Timeout = 3600
Idle-Timeout = 60
Acct-Interim-Interval = 120

**Billing Plans – Time Based**
Go to Billing Tab> Select Plans > New Plan
1. Enter Plan Information details from Plan Name to Plan Active
2. Enter Time Settings details
3. Select Profile from the drop-down

Then in adding users/batch users, I select my created Plan and Group in their appropriate dropdown.
This did not works for me before, but now is.

pihomeserver, 2017-06-08:
I confirm that it works. The point that should be improved is that the user is disconnected but when he tries to access again a website, no "user friendly" message is provided. Only a time out message. One more point to be improved

closed 2017-06-08 by pihomeserver