pihomeserver/Kupiki-Hotspot-Script #24: Ethernet connection goes down after line 192 of the script

mdorazio, 2017-04-06:
I attempted to run the Pi-Hotspot script this morning and got stuck after line 192.  After the message "Restarting network service to take IP forwarding into account" the ethernet connection no longer has internet connectivity and the rest of the script fails due to lack of connectivity.

eth0 shows it is configured (at 192.168.10.104/24), wlan0 is not associated.  Ethernet port lights are blinking and my router shows the Pi is connected and has the same IP address.  There's simply no internet access.

Config changes from original settings: Only changed hotspot name and all passwords

Things tried: restarting the Pi had no effect

pihomeserver, 2017-04-06:
Hi
Can you please check your interface configuration ?
What distro did you used ?

mdorazio, 2017-04-07:
Sorry for the delayed response.
I'm running stock Raspbian version 8

Here's the ifconfig output:
eth0      Link encap:Ethernet  HWaddr b8:27:eb:6f:bc:19  
          inet addr:192.168.10.104  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::7161:f0ba:d8d1:61dd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:45744 errors:0 dropped:0 overruns:0 frame:0
          TX packets:329 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:7707125 (7.3 MiB)  TX bytes:47207 (46.1 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6616 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6616 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:535680 (523.1 KiB)  TX bytes:535680 (523.1 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:3a:e9:4c  
          inet addr:192.168.10.1  Bcast:192.168.10.255  Mask:255.255.255.0
          inet6 addr: fe80::898e:3433:852e:5dbc/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:67278 errors:0 dropped:67278 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:28589142 (27.2 MiB)  TX bytes:0 (0.0 B)

pihomeserver, 2017-04-08:
Ethernet interface is on the same network than the hotspot 192.168.10.0
Try to set your hotspot on another network. You can change it innthe first lines of the script

mdorazio, 2017-04-10:
Thanks!  Changing the IP in the config and re-running the script from a clean raspbian install fixed everything.

closed 2017-04-10 by mdorazio