pihomeserver/Kupiki-Hotspot-Script #25: Issue on IPv4 network

gagan0123, 2017-04-17:
You need to execute this command before installing via the script on Rasbian Lite
```echo 'Acquire::ForceIPv4 "true";' | sudo tee /etc/apt/apt.conf.d/99force-ipv4```
when you are on IPv4 network, otherwise Rasberry will try to pull some packages from IPv6 network and script will fail.

pihomeserver, 2017-04-17:
Hi
What distro are you using ? Minibian and Raspbian lite do not need this command and the script has been validated on them without that

gagan0123, 2017-04-17:
My router is assigning IPv6 addresses as well as IPv4 addresses, so when Raspbian Lite tries to fetch apt packages, I'm getting 
```connect (101: Network is unreachable) [IP: 2001:e10:6840:3::70 80]```
the internet I'm having does not support IPv6, only internally IPv6 is being used among devices.

pihomeserver, 2017-04-17:
Ok thank you.
Can you please open a pull request with the update ?

closed 2017-06-08 by pihomeserver