pihomeserver/Kupiki-Hotspot-Script #26: Installation fails

Claymobil, 2017-04-17:
Hi,

during the process of installation, it doesn't seem to go al the way to the end because in the logs, i can find these lines:

`Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-common armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-cli armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Get:33 http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-json armhf 1.3.6-1 [16.9 kB]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php-pear all 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Get:34 http://mirrordirector.raspbian.org/raspbian/ jessie/main libapparmor1 armhf 2.9.0-3 [58.1 kB]
Get:35 http://mirrordirector.raspbian.org/raspbian/ jessie/main nginx all 1.6.2-5+deb8u4 [72.6 kB]
Get:36 http://mirrordirector.raspbian.org/raspbian/ jessie/main php-db all 1.7.14-3 [73.0 kB]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-fpm armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-gd armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-mysql armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Err http://mirrordirector.raspbian.org/raspbian/ jessie/main php5-readline armhf 5.6.29+dfsg-0+deb8u1
  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]
Get:37 http://mirrordirector.raspbian.org/raspbian/ jessie/main libgd2-xpm-dev armhf 2.1.0-5+deb8u9 [1,236 B]
Fetched 9,015 kB in 6s (1,314 kB/s)
E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-common_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-cli_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php-pear_5.6.29+dfsg-0+deb8u1_all.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-fpm_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-gd_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-mysql_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Failed to fetch http://mirrordirector.raspbian.org/raspbian/pool/main/p/php5/php5-readline_5.6.29+dfsg-0+deb8u1_armhf.deb  404  Not Found [IP: 2a00:1098:0:80:1000:75:0:3 80]

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
** 
** Something went wrong with the last command. Please check the log file
** 

Does these message can be resolved ?

thanks !

pihomeserver, 2017-04-17:
Hi
As said at the end of the log there is an issue with apt-get to fetch archives. It's not related to the script. Fix that and run the script again

pihomeserver, 2017-04-17:
I see ipv6 in your log. Look at suggestion in #24

Claymobil, 2017-04-25:
Oh.. I totally forgot to tell that i managed to succeed the installation with ytour help. thanks,

Now I would like to create a captive portal where the users could create themselves their account with their mail adress, is it possible ?

thanks

pihomeserver, 2017-04-25:
Good news !

Shoul be possible using the register function within coovachili. But never tried. If you succeed please share your progress ;-) not sure how to link both. May be you will need a mail provider like mailgun to send register email

pihomeserver, 2017-05-22:
Installation done successfully

closed 2017-05-22 by pihomeserver