pihomeserver/Kupiki-Hotspot-Script #28: Enhancement for modifiable login/registratin page

gagan0123, 2017-04-17:
I checked out how things are working in this script, its a nice thing you've made, but mostly people do this so that they can have their own custom pages for login and session management.
The defalt ones that comes with CoovaChilli are not quite good and if you want to modify them, its a PIA.

I think I have a solution to handle this problem, also removing the need of doloradius. Both things can be done in a single site. Since I'm a WordPress developer, thinking of developing a WordPress plugin to deal with FreeRadius database for things like, User Management(just a sync of users in WP with freeradius), Analytics, and Session management, as well as write a custom responsive theme which will allow users to select colors, change fonts, and text, upload/modify logo, and anything that goes on the frontend of the login/registration page.

Still experimenting with stuff and could take a lot of time though but end product would be awesome, might need your help with freeradius and CoovaChilli related things.

Also, we can use some sort of SMS based API to send activation codes or maybe emails. Its quite easy thing to do with WordPress.

pihomeserver, 2017-04-26:
That's currently in the next steps, like making a daloradius-like more light but also more modern
Any help is welcome to speed up the process :-)

reigelgallarde, 2017-06-06:
@gagan0123 , any progress on this?

would really be great if we can skip how daloradius is doing things.

Like should have merged chilli and daloradius. Currently we have to server. The chilli which is for the users and daloradius for the admin. But I know it's possible to just merge the two idea in one server.

So that we can have set of roles for each user. Where a **user role** for an account can only access the internet (like chilli). And another **admin role** for an account that can manage the user accounts (like daloradius). All in one login page.

I found some [discussions here](http://www.chillispot.org/chilliforum/topic32-chillispot-configuration-hotspotlogin.html) that they have converted chilli into php. I have not tested it but if that works, it can be our first step.

pihomeserver, 2017-06-29:
@gagan0123 @reigelgallarde from now the portal used is in HTML/JS/CSS exposed by the nginx instance used by freeradius (but different server block)
You can then customize it or write your own in php if you want

pihomeserver, 2017-07-06:
@gagan0123 any free time to start with us this part of the project ?