pihomeserver/Kupiki-Hotspot-Script #29: How to modify the captive portal splash screen and sign-in process?

DavSanchez, 2017-04-26:
Hello! I tried this script two days ago and it works perfectly with just a few minor changes on my side.

I have an idea for a project that involves the clients obtaining internet access not by intentionally signing in (I mean typing username and password), but only when the server gets some info from them through the browser, say location data and a timestamp to do some statistics. The server would renew the session in intervals of 30 min, at the end of which new location and timestamp information is collected from the clients, disconnecting them if it fails, but currently I don't know were to begin to modify the captive portal/splash screen to make this possible and then how to make the clients automatically sign in once the data required is collected (via some HTML/JavaScript I suppose).

I think I need more information on how the splash screen HTML is generated by the software to begin with. Any hints on where should I look into? Thanks in advance!

pihomeserver, 2017-04-26:
It seems that you can change the way CoovaChilli is handling the front to authenticate users
There are some exmaples somewhere in the installation folder of daloradius : /usr/share/nginx/html/daloradius/contrib/

You can use PHP, JS etc. 
I never tried yet but as said in #28, it's in the next steps list

DavSanchez, 2017-04-26:
Thanks for the quick answer! I'll take a look and see what I can get.

pihomeserver, 2017-06-14:
I was able to perform a basic test, getting the free sign in portal in Daloradius, put it in the nguni root folder and set the url of uamhomepage in /etc/chilli/main.conf

You can create a custom landing page
You will always need a user/password to log in (the portal will create a random user/password before redirecting to the standard login page)

DavSanchez, 2017-06-18:
Hello again!

I have been working on a Node.js app to serve the landing page and give the credentials to the clients, I want to put the app in the nginx html folder (or another location with permissions to write files) and edit the nginx configuration file to allow other connections to it (the same way it works with daloradius).

I guess I only have to add another location directive to the /etc/nginx/sites-available/default file and it won't mess with the daloradius location, but the solutions I have checked on the internet have different ways to do this and I'm afraid to mess with all the previous configuration. I will be doing some tests for the next couple of days and I'll see what I come up with.

Thanks!

pihomeserver, 2017-06-18:
HelloFor now on my side i try to use and implement the chillilibrary.js but have some unexpected issues with the json managementHope that you will find a solution i would like to use nodejs too at the end
-------- Message d'origine --------De : DavSanchez <notifications@github.com> Date : 18/06/2017  22:31  (GMT+01:00) À : pihomeserver/Pi-Hotspot-Script <Pi-Hotspot-Script@noreply.github.com> Cc : Pi Home Server <pihomeserver@gmail.com>, Comment <comment@noreply.github.com> Objet : Re: [pihomeserver/Pi-Hotspot-Script] How to modify the captive portal splash screen and sign-in process? (#29) 
Hello again!
I have been working on a Node.js app to serve the landing page and give the credentials to the clients, I want to put the app in the nginx html folder (or another location with permissions to write files) and edit the nginx configuration file to allow other connections to it (the same way it works with daloradius).
I guess I only have to add another location directive to the /etc/nginx/sites-available/default file and it won't mess with the daloradius location, but the solutions I have checked on the internet have different ways to do this and I'm afraid to mess with all the previous configuration. I will be doing some tests for the next couple of days and I'll see what I come up with.
Thanks!

DavSanchez, 2017-06-21:
Just in case your JSON problems are the same I ran into today, you should know that in order for the login process to work correctly with ChilliLibrary.js you need to disable the uamsecret (that is, leaving blank the HS_UAMSECRET attribute in the /etc/chilli/config file). I just did that and the whole thing worked on reboot!

You can check [my repository](https://github.com/DavSanchez/CaptivePortal) if you want to see what I have been doing with it. It's still incomplete with some problems and without a useful readme file yet, but it gives internet access correctly with credentials given from a server. Feedback appreciated!

Thanks for your help!

DavSanchez, 2017-06-21:
I have recommended and linked your repository on mine's Readme file. If you prefer that I don't do this just let me know and I'll remove it!

pihomeserver, 2017-06-21:
Thanks for the advice. I saw that too and the test was in progress (i made a mistake while trying to use the git repo of daloradius, and deleted all)
I will test your files

DavSanchez, 2017-06-27:
Apparently it's not needed! I checked that the default UAMHOMEPAGE redirects to the UAMFORMAT provided after its timeout, so I left it untouched for the time being at least...

pihomeserver, 2017-06-27:
Thank you 
I was able to implement the authentication in a single page but got some issue with the iptables rules to expose the page without opening the whole network

pihomeserver, 2017-06-29:
@DavSanchez from now the portal used by the hotspot is the one in the repository pihomeserver/Kupiki-Hotspot-Portal
It has less functions than yours but could be easy to update and customize
Thank you again for your help

DavSanchez, 2017-06-29:
I'll check it out. Thanks to you too!

emperor443, 2017-09-21:
i used the ready-to-use image with this with this link https://drive.google.com/file/d/0B5CzDtjmXSaySVNPZ1A1VkYtVFk/view?usp=sharing everything boot up successfully but during the login i had a problem anytime i used this Credentials are user : pi / password : pihotspot  it gives me wrong password..please kindly update me with the password,,thanks

closed 2018-03-03 by pihomeserver

outbackdingo, 2018-07-27:
so let me ask, to clarify whos repo can i use to test a login solution  that doesnt require a username / password froom the user, we just want them to click a ACCEPT TOS button and be authenticated.

pihomeserver, 2018-07-27:
Maybe there is an option in coova but else you can create a dummy user in freeradius and then hardcode these credentials in the captive portal source

vanrion, 2018-08-08:
Use MAC authentication for Auto-Login.

pihomeserver, 2018-08-08:
@vanrion : are you sure about that ? I have in mind that MAC authentication is checking a user with the MAC address with a default password. If it matches then you will be logged else you will be redirected to the standard login/password form