pihomeserver/Kupiki-Hotspot-Script #31: WPA2 Encryption

bbrowder, 2017-05-22:
First of all, thank you for providing this script.  I have found it extremely useful.

I am wondering, however, if it is possible to encrypt the wifi traffic using WPA2, but still use the captive portal login?

pihomeserver, 2017-05-22:
Hello,
Sure you can with the hostapd configuration file (search for wpa2 in the configuration file https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf) but it means that you have to give the network password before the user can reach the login page and i think that's not the purpose of a captive portal.

bbrowder, 2017-05-22:
You're 100% correct, and that is what I would like to avoid.  There is a CoovaChilli setting called WPAGUESTS that seems like it might allow me to accomplish this, but I haven't been able to get it working.  I didn't know if you had any experience with it.

https://coova.github.io/CoovaChilli/WPACaptivePortal/