pihomeserver/Kupiki-Hotspot-Script #35: Daloradius has github.

reigelgallarde, 2017-06-19:
I found this while searching the web for things.
https://github.com/lirantal/daloradius

It looks more updated than that of sourceforge.
I also make a pull on that for the duplicate entry of `sqlcounter noresetcounter`
related to the script pihotspot.sh lines 466 - 468

maybe we can use github version?
I have not yet tried it though. I might this coming days if I'm not busy.

pihomeserver, 2017-06-20:
Thank you for the information. I installed it in place of the previous version and it seems to work as expected. I will update the script and then remove the sed that removes the duplicate entry

pihomeserver, 2017-06-21:
It's strange, the script does not work anymore with a dhcp error on the coova start 
```
chilli[16062]: (Re)processing options [/var/run/chilli.16062.cfg.bin]
chilli[16063]: running chilli_opt on /var/run/chilli.16062.cfg.bin
chilli[16062]: PID 16062 rereading binary file /var/run/chilli.16062.cfg.bin
chilli[16062]: Loading modules
chilli[16062]: PID 16062 reloaded binary options file
chilli[16062]: CoovaChilli 1.4. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-2012 David Bird (Coova Technologies). Licensed under GPL. See http://coova.github.io/ for details.
chilli[16062]: chilli_main(7416): clock realtime sec 1498076805 nsec 55719613
chilli[16062]: chilli_main(7423): clock monotonic sec 1126 nsec 63600043
chilli[16062]: TX queue length set to 100
chilli[16062]: Running /etc/chilli/up.sh
chilli[16062]: Hashlog 5 27 32
chilli[16062]: removing ip address from wlan0
chilli[16062]: 92 setsockopt(s=5, level=6, optname=1, optlen=4) failed
chilli[16062]: Failed to create dhcp listener on wlan0
```

reigelgallarde, 2017-06-22:
@pihomeserver can you create a another branch (a test branch or something) for this new script? I wanna test it too...

pihomeserver, 2017-06-22:
@reigelgallarde done with the branch called 'git_daloradius'

pihomeserver, 2017-06-22:
Solution found within the issues of Coova coova/coova-chilli#368

pihomeserver, 2017-06-22:
Included now in the new script version

closed 2017-06-22 by pihomeserver