pihomeserver/Kupiki-Hotspot-Script #37: JSON request time out

vanrion, 2017-06-28:
there's was an error when redirecting in Portal and also  when trying to login with this new Kupiki-Hotspot..

pihomeserver, 2017-06-28:
Can you please check at the html content of /usr/share/nginx/portal/index.html and be sure that the IP of the coova server is well set ?
`chilliController.host = "192.168.10.1";`

pihomeserver, 2017-06-28:
I have just tried from scratch with default parameters within the script and i do not have the issue

vanrion, 2017-06-28:
yes it is. maybe i should try another fresh installation.

vanrion, 2017-06-28:
it redirect in this ip 192.168.0.34:9000/?index.html...something like that, which is my raspberry pi local IP.

pihomeserver, 2017-06-28:
Yes that could be the reason even if i tried to not hardcode IP address
The IP in the HTML should be the WAN interface with the port 9000 (not the hotspot IP)
You could reach it from any browser on the same network

vanrion, 2017-06-29:
ok, i just do fresh installation with default parameter and it's working.. :+1: 
Now, i wanted to add another page for Time Remaining. any idea Sir?.

pihomeserver, 2017-06-29:
If you use http://status in the  location bar you should have some useful information
i plan to use the same page to display that information if the system detects that you are connected

closed 2017-06-30 by pihomeserver

vanrion, 2017-07-08:
@pihomeserver  I have this error again. 

[image]

pihomeserver, 2017-07-09:
Arf that's strange. Maybe it's due to the custom ip. Did you try to run coova and freeradius manualy to see possible error message ?
I will try to run the script on a frresh install. Can you please provide your custom parameters to use the same ?