pihomeserver/Kupiki-Hotspot-Script #41: Reinstall issues

mclaassen, 2017-07-05:
Great project!
Since I had to reinstall due to issues I noted that reinstall failed due to some existing git installs and symbolic link.
The portal did not work initially but I figured out that the XXX replacement in portal.js was not happening

#Changed this to the pihotspot.sh code so that reinstall is also functioning, especially chilli files were not overwritten:
#Making sure old configs are deleted:
dpkg --purge coova-chilli
dpkg --purge haserl
dpkg --purge hostapd

#Modifying line below to enable overwrite symlink
#from:
execute_command "ln -s /etc/nginx/sites-available/portal /etc/nginx/sites-enabled/portal" true "Activating portal website"
#to:
execute_command "ln -sfT /etc/nginx/sites-available/portal /etc/nginx/sites-enabled/portal" true "Activating portal website"

#Modifying the portal and daloradius git install so that old files are deleted first:
execute_command "cd /usr/share/nginx/html/ && rm -rf daloradius && git clone $DALORADIUS_ARCHIVE daloradius" true "Cloning daloradius project"
execute_command "cd /usr/share/nginx && rm -rf portal && git clone $HOTSPOTPORTAL_ARCHIVE portal" true "Cloning Pi Hotspot portal project"

#Modified replace line XXX to changes in portal.js as there the hotspot should be configured
sed -i "s/XXXXXX/$HOTSPOT_IP/" /usr/share/nginx/portal/js/portal.js

pihomeserver, 2017-07-05:
Thanks for suggestion. I will create a branch to test all of that ;-)

mclaassen, 2017-07-10:
Two more issues I found when testing. Don't know if it is applicable for everyone but that's better for you to decide :-)
Since DHCP is (often) being used on the WAN side of the router I changed the HS_UAMFORMAT parameter in the script to the Hotspot IP address as below.
Furthermore I experienced problems getting to the landingpage. Solved by including port 9000 to allowed ports. 

display_message "Updating UAMFORMAT"
sed -i "s/^HS_UAMFORMAT=.*$/HS_UAMFORMAT=http:\/\/$HOTSPOT_IP:9000/g" /etc/chilli/defaults
check_returned_code $?

display_message "Add firewall allowed port"
sed -i '150iHS_TCP_PORTS="9000"' /etc/chilli/defaults
check_returned_code $?

pihomeserver, 2017-07-10:
Thank you @mclaassen 
Have to integrate all of that but with a dynamic port instead of 9000 hardcoded

pihomeserver, 2017-07-10:
@mclaassen if you want to change the UAMFORMAT then the nginxi configuration has also to be updated (and add escape characters to the http://)

mclaassen, 2017-07-12:
Yes indeed.. 
Hmm, for some reason that got wrong in this comment as I have it correct in my script.

pihomeserver, 2017-07-13:
Included in v1.1

closed 2017-07-13 by pihomeserver

mclaassen, 2017-07-14:
Perfect. I tested the install and it was fully correct from start!
I have one more issue when using iOS. I'll open a new issue for that..