pihomeserver/Kupiki-Hotspot-Script #43: How to configure daloradius to build a multispot network

pihomeserver, 2017-07-06:
For those who want to use Kupiki-Hotspot in a large area that could not be covered by a single Raspberry Pi

vanrion, 2017-07-07:
Repeater Mode using CPE it might work?

pihomeserver, 2017-07-07:
@vanrion may be my issue is not well formated. It seems that with daloradius you can build a hotspot network with a centralization of the management on one node. I would like to investiguate how to configure Kupiki-Hotspot to act as a node like configuring the use of a remote instance of freeradius (is that enought ?)

mclaassen, 2017-07-10:
I was working on that same solution too.
When having multiple hotspots serving the same network it's key to take the freeradius configuration out of the configuration of your Pi and place it in a central location. That would mean that your pi only has the chilli and webserver configuration and that radius, dalo and mysql are running centrally. User management is central at that time. Each pi node has it's own identity and users can roam from one hotspot to the other (they'll have to relogin though)

pihomeserver, 2017-07-10:
That's exactly what i have in mind. Using the same script to configure :
- A all-in-one hotspot
- A node hotspot
- A "backend" hotspot

Help is welcome :-)

mclaassen, 2017-07-27:
I have now a configuration where I have 3 AP's running all connecting to the same instance for authentication. Since my AP's are connected via 4G dongles (so no direct access) I also needed a way to be able to manage them. Therefore I used OpenVPN tunnels from the AP's to the authentication server. 
The other way (maybe better) would be by using radsec authentication but since I (as stated before) needed remote management I chose the OpenVpn solution.
Let me know if you need more details!

pihomeserver, 2017-07-27:
About configuration of all, except the openvpn part, i guess that you updated the coova configration to not connect in local to freeradius but the remote (and unique) instance

mclaassen, 2017-07-28:
That's correct. There is not much to configure actually. Since all my AP's are coming from the same subnet the only thing i had to do is adding the subnet to the list of NAS configurations in radius.

denis-ev, 2017-08-04:
Hey, I recently used your script to setup a hotspot for a friend and was just checking the issues. I've read that you are using openvpn for the secure connection between the APs. I just wanted to throw another option for VPN Access in, Tinc (https://www.digitalocean.com/community/tutorials/how-to-install-tinc-and-set-up-a-basic-vpn-on-ubuntu-14-04). It's a VPN Mesh, so every AP could talk with every other. Hope this can help, to create a fantastic Hotspot-Script.

Digitalocean link: https://m.do.co/c/b55f5fb48aca

Best regards,

pihomeserver, 2017-08-04:
@denis-ev thank you for the share. That could be interesting to test it and integrate it to the script !

_Sent from my Samsung SM-A510F using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_