pihomeserver/Kupiki-Hotspot-Script #44: Add HTTPS support to front web portal

pihomeserver, 2017-07-07:
Could be interesting to add (force ?) HTTPS support for web portal with SSL certificate signed from letsencrypt. 
Adding also the renewal of the certificate to not get blocked after 3 months

Seems to be required to not allow PAP authentication occurs in clear over the network

mclaassen, 2017-07-10:
That's a good point. Especially when referring to security in issue #43 certificates can be useful when using a radsec implementation in combination with freeradius and chilli

pihomeserver, 2017-07-14:
Letsencrypt (and other CA) does not allow signed certificate for localhost. Self signed certificate have to be used but then an alert will be raised

mclaassen, 2017-07-14:
in that case you'd need a dns name assigned to that hotspot and request a certificate for that. The name would be fixed then though which also is not very useful in case you want your own dns name assigned to it.
Only option i see is that you install the certificate and dns name that need to be provided by the one installing the script. So they need to assign the name to the hotspot and request the certificate. Maybe not worth the effort...

closed 2018-03-30 by pihomeserver