pihomeserver/Kupiki-Hotspot-Script #45: HotSpot Chambre d'hotes

neosoda, 2017-07-08:
Salut ! 

Merci pour les tutos et informations hyper intéressantes !

J'ai besoin d'installer un hotspot Wifi dans une seule chambre d'hôtes, de façon sécurisée et différente de la connexion privée de la propriétaire. Jusque là, ta solution semble parfaite.

J'ai cependant une question, étant complètement novice dans ce genre de projet avec raspberry et linux...
Cette solution propose-t-elle une sauvegarde des fichiers logs ? Etant donné que la loi impose une sauvegarde de 12 mois de ces fameux logs...Une solution ? Est-ce intégrer ?

Comment le débutant que je suis (linux et raspberry) peut faire pour mettre en place ce système de HotSpot.

Merci

pihomeserver, 2017-07-09:
Bonjour
Dans les points encore ouverts, je dois travailler sur l'integration de loggsi pour une demande similaire.

En theorrie, si vous avez un raspberry avec une raspbian lite installée il suffit de lancer le script. Toujours dans les points ouverts je dois faire une image basique pour simplifier l'installation mais le parametrage post installation sera plus complexe

pihomeserver, 2017-07-21:
Image disponible (voir README)

pihomeserver, 2018-02-04:
Logs are now included in the 1.5 release of the script

closed 2018-02-04 by pihomeserver