pihomeserver/Kupiki-Hotspot-Script #46: No ''done" button on iOS

mclaassen, 2017-07-14:
When using the captive portal via an iOS device the popup landing page of iOS never shows you the 'Done' link on top right after success login. It remains 'Cancel' although you are logged in already.
By enabling the Redirect function (line 75) in the portal.js file that problem is gone. 
I know it's only a cosmetic problem cause if you click the home button you are connected to internet. The Done link on top is just nicer in this case and if you happen to click the Cancel link you are disconnected again.
Having a configurable redirect page (line 19) would also be nice in that case

pihomeserver, 2017-07-14:
Exiting bug on MacOs too.

pihomeserver, 2017-07-15:
The cancel button will change to Done on successful communication with the captive.apple.com. If the device is not able reach the apple server, the done button will not appear.

mclaassen, 2017-07-19:
Yes I know that that is the idea but when I remove the redirect call from the js file the captive portal does not change that button to 'Done'. It remains 'Cancel' forever and only when I press cancel I get Disconnected. Adding the redirect ni the js file gives me a much better user experience. I think the reason for that is that the login screen does not really doing a request for a new page. The redirect forces a new page causing the device doing a new call to the captive.apple.com to check whether internet is available..

pihomeserver, 2017-07-19:
Maybe instead a redirect that is not really "clean" (IMHO) we should try to do an AJAX call like to a website to force the OS to recheck the apple website ?

mclaassen, 2017-07-19:
That would even be better. A redirect is a 'nice to have' in the configuration but should not be a requirement indeed

pihomeserver, 2017-07-21:
Web portal updated in v0.3 to reload the homepage if the authentication is a success then the button 'Done' is displayed

closed 2017-07-21 by pihomeserver