pihomeserver/Kupiki-Hotspot-Script #48: Extremely slow access

chuckgenco, 2017-07-27:
I've done the install, all looks good. No errors but I have extremely slow access and just getting the captive splash/login takes forever to come up and even then can fail. If I succeed, hitting fast.com will give me something like 2kbs.

Raspberry Pi 3 with the built in wifi using the latest rasbian jessie lite on ethernet
Version:July 2017
Release date:2017-07-05
Kernel version:4.9

I'm at a loss here. Any ideas?

pihomeserver, 2017-07-27:
Hello
I do not have this issue and i use the exact same configuration. How far from the pi are you ? Did you try the wifi with raspbian and not this config ?

chuckgenco, 2017-07-27:
I'm 2 ft away from the pi. Tried connecting with multiple android and iPhones plus mac laptop. By other config, you mean an older version of the os or pi or using a dongle? I did try with the lite and full desktop versions. The pi has no issue getting its network connection from wifi before setup. To do the setup I connect the pi via Ethernet. I've tried the install after connecting to wifi and never connecting to wifi. I can see if I can get an older version of the is or another pi.

chuckgenco, 2017-07-27:
The signal indicator on the devices is strong

chuckgenco, 2017-07-27:
with ssh I do get a message after commands
"sudo: unable to resolve host pihotspot"
and I seem to have slightly better luck when the pi is headless

pihomeserver, 2017-07-27:
Ok can you please check that your hostname (/etc/hostname) is also defined in the hosts list (/etc/hosts) ? That should solve your sudo issue

chuckgenco, 2017-07-27:
Actually I made that change about 15 minutes ago after a google search. That did solve that.
I still have the other issue

pihomeserver, 2017-07-27:
Can you please share /etc/hostname and /etc/hosts content ?
Did you made a reboot after the installation ?

chuckgenco, 2017-07-27:
Yeah, it's been rebooted. Installing it on an earlier version of rasbian didn't help.

/etc/hostname
pihotspot

/etc/hosts
127.0.0.1       localhost
::1             localhost ip6-localhost ip6-loopback
ff02::1         ip6-allnodes
ff02::2         ip6-allrouters

127.0.1.1       raspberrypi
127.0.1.1       pihotspot

pihomeserver, 2017-07-28:
Not related to raspbian as i used the same version.
Did you try to download from your pi a big file to check your internet connection ?
Else take the previous version of the script and use it on a fresh raspbian (the latest is ok)

chuckgenco, 2017-07-28:
By download to the Pi a big file do you mean before the setup? I have no problems just using the WiFi connection as my Pi network connection.
Install is done via ethernet on a fresh version of raspbian never connected via WiFi.
Here is what happens a bit more exactly.

1. I put a version of rasbian-lite on an SD card (I've tried multiple SD cards just to eliminate that as a possibility, however unlikely)
2. I install get, clone your repo, follow the instructions (the first time I edited the name but after that left the script alone, (I've done this it via ssh and with a keyboard an monitor... I can see absolutely no reason this should make a difference but again, just in case)
3. reboot.
4. I go to my Mac I add a user. (I've no problems hitting the admin pages over ethernet)
5. The indication of the network signal is strong. I then try and join the network on either an iPhone, an android or my mac laptop.

Sometimes I can't even initially join. Most often the portal page doesn't even come up or doesn't load. Sometimes on the iPhone, I get a system dialog asking for a password. What password it is asking for I've no idea. It doesn't ask for a user name. I dismiss it. Sometimes the portal page doesn't even come up and if it does, not all the assets come through. If i can manage to login, the confirmation page doesn't really load completely.
If I manage to log in, 99.9% of the time even hitting a light page won't load. the other .1% of the time it loads but slowly.
Once there was great performance for a time and then dropped to nothing.

I really appreciate your help. This is the perfect solution for my needs.

I will be away until Monday so won't be able to try your suggestion for a time.

I doubt it is the issue but to make sure I am ordering a new Pi. (seems unlikely the issue, but I'm covering all the bases.)
I will try your suggestion of the current script on a new card. Then if there are issues, the previous version of your script.

Again, thank you for your help.

chuckgenco, 2017-08-01:
I think we are good. After trying the options above, we got a 50ft ethernet cable and discovered it must be some kind of wifi interference. Thanks for all your your help and thanks for the script!

closed 2017-08-02 by pihomeserver