pihomeserver/Kupiki-Hotspot-Script #50: Error on ./pihotspot.sh

Jmunapo, 2017-08-17:
Am having trouble, am using ssh to install but i get this error -> Something went wrong with the last command. Please check the log file

Help please

pihomeserver, 2017-08-17:
Sure i can help but please follow instructions first : the message asks you to check the log file. Did you read it ?

Jmunapo, 2017-08-17:
Thanks for replying, am using SSH so i cldnt read it, let me find a monitor

Jmunapo, 2017-08-17:
There is no  Log file, I'm kinda newbie here, can you please help or recommend any alternative

pihomeserver, 2017-08-17:
I recommend to read carefully what is displayed since the launchof the script and in the readme on the landing page on github :

`A log file named pihotspot.log will be created in the folder /var/log. Open a new session on the Pi and run the following command tail -f /var/log/pihotspot.log`

So go in the folder, cat the log file and check what is the error at the end of the script
Also : what distribution did you used ? Seems that Raspbian is not working due to extra packages (to be confirmed), you have to use Raspbian Lite

africanalex, 2017-08-18:
I am getting this error too on a fresh install of raspbian stretch lite, my error log is below, it is saying mysql is not installed. Is mysql listed in the dependencies?

```
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
::: Updating the system hostname
::: Adding MySql password
::: Adding MySql password (confirmation)
::: Getting WAN IP of the Raspberry Pi (for daloradius access)
::: Starting MySql service
Failed to restart mysql.service: Unit mysql.service not found.
::: 
::: Something went wrong with the last command. Please check the log file
::: 
```

pihomeserver, 2017-08-18:
@africanalex can you please run manually the following command `service mysql restart`

If ok then run the script again else try running the command `mysql` and check if a message is displayed

africanalex, 2017-08-18:
@pihomeserver 

I ran: 
sudo service mysqld start
and got:
Failed to start mysqld.service: Unit mysqld.service not found.

so I installed mysql-server and it got past the problem, getting a new error with naming of eth0 but that is a separate issue

pihomeserver, 2017-08-18:
Really strange as mysql-server is part of the dependencies to install

fabcal24, 2017-08-21:
mysql is deprecated and now apt install only mariadb instead. mariadb is fully compatible with mysql's databases

pihomeserver, 2017-08-21:
@fabcal24 not sure that's the reason as the mysql-server is still available in the 8.0 trunk from http://packages.debian.org :
> Package mysql-server
> stretch (stable) (database): MySQL database server binaries and system database setup [transitional] 
> 5.5.9999+default: amd64 arm64 armel armhf i386 mips mips64el mipsel ppc64el s390x

pihomeserver, 2017-08-21:
@mclaassen Thanks for the tip. Have to test that (too much work on the new admin interface ...)

mclaassen, 2017-08-21:
I'm checking it too.. I also just figured out that debian Stretch comes with freeradius 3 which requires a different setup :-(
Anyway. I'm checking what more needs to be changed and keep you updated..

fabcal24, 2017-08-21:
I suggest dowload previous version (jessie) here: http://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/
but change mysql-server with mariadb-server. I try with mariadb and work fine

pihomeserver, 2017-08-21:
I will redo a full test next days to control all. I missed the Raspbian upgrade and the regression related to it ;-)

fabcal24, 2017-08-21:
Dont' forget: in strechy all lan devices named enp0s0 et similar and needed some modifications for change to canonical name like eth0 or wlan0. Also net-tools not installed by default and if needed, must to be added in script for installation. In strecthy, default tool is ip

fabcal24, 2017-08-21:
pihomeserver, if you want I can help you with my little contribute. If you want, you can contact me in private email

pihomeserver, 2017-08-27:
Corrected in v1.4
Please look at CHANGELOG for changes. Any additional testers are welcome :-)

fabcal24, 2017-08-27:
Good, today I try this!

ahgpeu, 2017-08-30:
I have an error while installing.
Last distrib doesnt work...
`Cloning into 'coova-chilli'...
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
:::
::: Something went wrong with the last command. Please check the log file
:::
`

pihomeserver, 2017-08-30:
@ahgpeu : last distro = Raspbian Stretch **Lite** ? I did not get this last week end. And it's related to git, not the script (you can search for the error)

Can you please go in /tmp and type the following command : 
```
git clone https://github.com/coova/coova-chilli.git coova-chilli
```

ahgpeu, 2017-08-30:
Last distrib of script and last distrib of Raspbian Stretch Lite.
Cloning coova-chilli with command works well:
`
Cloning into 'coova-chilli'...
remote: Counting objects: 7858, done.
remote: Total 7858 (delta 0), reused 0 (delta 0), pack-reused 7858
Receiving objects: 100% (7858/7858), 3.54 MiB | 239.00 KiB/s, done.
Resolving deltas: 100% (6329/6329), done
`

pihomeserver, 2017-08-30:
Strange as it's the executed command line by the script to clone coova. Can you rerun the script please ?

ahgpeu, 2017-08-30:
After rerun script works well. I have tried two times.
1. Install raspbian stretch lite.
2. apt-get update, apt-get upgrade, make udev rules to give normal names to interfaces. 
3. run script (recive error)
4. run script (works well)
But always AFTER I REBOOT it, I have error, when I try to enter with valid user/password to captive portal:
====
_The last contact with Controller failed. Error code=JSON request timed out (or reply is not valid)_

pihomeserver, 2017-08-30:
Ok the double error is strange

About other issue, please check that coova is running but also freeradius using `ps -ef` command. With the new Debian, getting freeradius started is not easy as before

ahgpeu, 2017-08-30:
Yes, problem was with freeradius.
After command:
/etc/init.d/freeradius start 
portal works well. So I aded string "/etc/init.d/freeradius start" to rc.local. I know, it is not correctly but now it works.

pihomeserver, 2017-08-30:
Good news thank you. I will update the script with the update of rc.local. For now it seems that there is an issue with that. Mysql is not started when freeradius is launched and then it crashes. The rc.local is the way ...

pihomeserver, 2017-09-04:
1.4.3 update with commands to add free radius in systemd as rc.local is not working as expected

closed 2017-09-04 by pihomeserver