pihomeserver/Kupiki-Hotspot-Script #53: Deleting and disabling user on daloradius

Jmunapo, 2017-08-21:
Now it works but am having trouble managing users, if i disable or delete them they keep accessing the internet is it the way it works or there is something i'm missing?

pihomeserver, 2017-08-21:
That's not the expected behavior. Can you please try to stop start coova after ? Maybe there is a kind of cache that has to be cleared once updated even if for creation there is no need ...

_Sent from my Samsung SM-A510F using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

pihomeserver, 2017-09-06:
Also you can try to disable the user, then disconnect it and then delete. Not sure about how the delete function works as i'm pretty sure that the information is not sent from freeradius to coova to update the access

closed 2017-12-27 by pihomeserver

petitours, 2018-11-24:
Bonjour
J'ai le même soucis. Je crée un user dans daloRadius ; je peux alors me connecter au hotspot avec ce user. mais si je désactive le user dans dalo, y compris en faisant un sudo reboot après (donc avec déconnexion forcée du user), le visiteur peut se reconnecter sans soucis via l'interface du hotspot, comme si son user était toujours actif.
Est ce un bug ou j'ai raté quelquechose ?
Merci

pihomeserver, 2018-11-24:
How long did you wait between disable and new login ? Did you try to disconnect the user once you disabled it ?
Once disable what are attributes in the database ?

petitours, 2018-11-25:
How long did you wait between disable and new login ? minutes and hours
Did you try to disconnect the user once you disabled it ? Yes, with or without forget about the wifi network.
Once disable what are attributes in the database ? which attribute in which database.

now nothing is working...I tried to restart the pihotspot.sh but it fails. 
I had to remove the webui repository and then restart the pihotspot.sh. The script was then successful but 
-daloraduis works
-the kupiki web form does not work anymore (the last contact with the controller (192.168.0.1:3990) failed. Error JSON request timeout....

short no way to reinstall, I am obliged to reinstall the entire raspberry (it will be done in 30min)

thanks

petitours, 2018-11-25:
I reinstalled everything with raspbian lite and rasp3B +
I get the same behavior
but I see that the log file is not perfect, some problems seem to be present but I do not understand them. 
[kupiki_hotspot.log](https://github.com/pihomeserver/Kupiki-Hotspot-Script/files/2613036/kupiki_hotspot.log)
here is the log file, does it help?

What I do after flashing Raspbian Lite ISO : 
sudo apt-get update && sudo apt-get upgrade
sudo passwd pi
sudo apt-get install git
git clone https://github.com/pihomeserver/Kupiki-Hotspot-Script.git
cd Kupiki-Hotspot-Script
nano pihotspot.sh (change mysql pass, SSID, active fileToBan)
sudo chmod +x pihotspot.sh && sudo ./pihotspot.sh
and then reboot by deconnecting power

Thanks

pihomeserver, 2018-11-25:
Hello
The script is not developt to be run more than one time. Else some configuration files will be corrupted

For the log file all seems to be ok with no error return code as it ends successfuly 

By the way i am pretty sure that it’s related to freeradius or coova configuration and not the script it self. Run freeradius and coova in debug mode in a terminal and check traces when the user is disconnected
I have no laptop for few days so i can’t check that my self