pihomeserver/Kupiki-Hotspot-Script #59: Help wanted - Administration portal pre-alpha testing

pihomeserver, 2017-09-11:
Hello,

I'm looking for 1 or 2 persons that would have enought time to help me in the testing of the administration portal that i'm creating for the hotspot.
For now don't expect so mush but with time functionalities should be :
- System management : 
 - basic monitoring like cpu, memory, uptime, services
 - basic administration : stop/start services, reboot, halt, update
- Hostpot management
 - Change hostapd configuration like hotspot name, driver, etc
 - Change coova configuration like ip range, dhcp, etc
- Hotspot portal management
 - Change background picture
 - Change few parameters like terms and condition
- Daloradius replacement
 - Provide a basic user management interface (create, edit, check, revoke, attributes, etc.)
 - Provide some other functions (vounchers, nas management)
 - Statistics to be defined

Technologies : Angular 1.6 + NodeJS
The first alpha has been tested on the Pi and seems to work great

Any volunteers ?

vanrion, 2017-09-27:
Hi, where can i download the image to test?

pihomeserver, 2017-09-27:
Can you please send me an email on my gmail mailbox (pihomeser at gmail dot com)

closed 2017-12-27 by pihomeserver

jeebee2013, 2018-01-15:
hi 
ok for testing
en plus on pourra échanger en Francais :-)

pihomeserver, 2018-01-16:
Salut
Peux tu m'envoyer un email (celui plus haut)? Pour éviter de polluer le flux