pihomeserver/Kupiki-Hotspot-Script #61: Web Managing Page

huseyinergen, 2017-09-29:
Hi everybody.. i installed (i used full image on this page) i logged pi/raspbian in cli. 

But i can not access via web page from another pc. What i make for this ? Please more explain how works this system

pihomeserver, 2017-09-29:
Please also explained what you made : did you create a user within daloradius ? did you log into the hotspot ? was the authentication solved ? 

Also i have to kick that full image that is using a so old release of the hotspot

huseyinergen, 2017-09-29:
i want login and manage hotspot. is there any web management page ? for example http://10.0.1.25/daloradius

pihomeserver, 2017-09-29:
No there is not for now (It's in progress and in testing) You have to either customize parameters of scripts (see first lines and README) to customize your SSID etc and use daloradius for users management
Once installed if you want to update the configuration you have to change configuration files manually.

The web interface release will help you to manage all configuration (including user management) in one place. But it will not be released within next weeks

morningstar74, 2017-10-05:
Hello.. Thanks for image.. I used full image install and i got same problem.. Can't login in hotspot..
I Haven't change anything in script.. I just want to login and idk user and password for login..

pihomeserver, 2017-10-05:
You have to use daloradius to create your users accounts. It's in the readme
-------- Message d'origine --------De : morningstar74 <notifications@github.com> Date : 05/10/2017  07:03  (GMT+01:00) À : pihomeserver/Kupiki-Hotspot-Script <Kupiki-Hotspot-Script@noreply.github.com> Cc : Pi Home Server <pihomeserver@gmail.com>, Comment <comment@noreply.github.com> Objet : Re: [pihomeserver/Kupiki-Hotspot-Script] Web Managing Page (#61) 
Hello.. Thanks for image.. I used full image install and i got same problem.. Can't login in hotspot..

I Haven't change anything in script.. I just want to login and idk user and password for login..

closed 2017-10-11 by pihomeserver