pihomeserver/Kupiki-Hotspot-Script #62: How can I allow domains for unauthenticated users?

andreaskasper, 2017-10-07:
Hi,
I wanna set my domain example.com to be accessible without having a user/password-Account.
Is it possible to whitelist some IPs/Domains in freeradius? Or allow some external URLs on the login-splash-screen like "[image]"

Thanks for your answer :-)
andreas

pihomeserver, 2017-10-08:
not sure how to do that. may e by adding a dedicated rule in the up script of the interface to always allow the destination without a credentials check before establishing the connexio

takatar, 2018-06-10:
Bonsoir,
Voilà ce qui a fonctionné pour moi:
- ouvrir le fichier /etc/chilli/defaults
- décommenter la ligne HS_UAMDOMAINS et saisir les domaines autorisés, .exemple.com,...
- reboot

Teko4444, 2018-12-12:
> Hi,
> I wanna set my domain example.com to be accessible without having a user/password-Account.
> Is it possible to whitelist some IPs/Domains in freeradius? Or allow some external URLs on the login-splash-screen like
> 
> `<img src="//otherdomain/importantimage.jpg"/>`
> 
> Thanks for your answer :-)
> andreas

same problem here and when i try @takatar solution some website like google work but others not working !! 
any ideas ??

pihomeserver, 2018-12-12:
@Teko4444 are you sure that you added all domains and there is no redirection ? Not an issue HTTP vs HTTPS ?

Teko4444, 2018-12-12:
@pihomeserver  it's not about the redirection , i want user to access my website firstly without authentication , if he want to use internet freely he should authenticate, i add .google.com to HS_UAMDOMAINS on /chilli/default its works fine but if i add my domain like .mydomain.com its not working i trying both http://mydomain.com and https://mydomain.com but its not show me my domain home page ! 
can you tell me what is the problem ?

pihomeserver, 2018-12-12:
Your domain is available from the Pi, did you look at logs, starting coova in debug mode and look at the trace. Also you can analyse the traffic on tun0 to see if request are rejected or are dropped

Teko4444, 2018-12-12:
ther is no directory  /var/log/pihotspot.log !! but every things works fine ! :D 
oky i will try coova in debug mode

pihomeserver, 2018-12-12:
Can you try to change the order in the parameter to see if it's not related to the parsing or something else ? 
I'm testing a new release and will check that after

Teko4444, 2018-12-12:
@pihomeserver i have delete all allowed domains except my domain then google by order but it still not working !!
can i do any thing till yr new release ?

thanks for help man

pihomeserver, 2018-12-12:
Do you stop start the service each time ? I don't know how coova manage the domains but maybe some iptables rules ? Then you need to stop start to update them

Teko4444, 2018-12-14:
@pihomeserver  i have made it just i put the domain without the . at first of it on /etc/chilli/default 
so it will be like  (.google.com , mydomain.com) that's all !!

pihomeserver, 2018-12-14:
So now it works without the dot ?

Teko4444, 2018-12-15:
@pihomeserver  yes without the dot its works for the domain only not for subdomains but it's fine for me.

pihomeserver, 2019-01-06:
Also : 

> All of those domains appear to be HTTPS domains with no HTTP counterparts.
CoovaChilli can't inspect the packet headers to determine the end
destination and determine if the packets should be forwarded because the
packets are encrypted, so they're dropped.

> Redirection

 / captive portal will never work with HTTPS-secured traffic -
not being able to figure out where the packets are going is kind of the
point of this protocol.

Source : https://www.brightonchilli.org.uk/pipermail/coovachilli/2017-June/000233.html