pihomeserver/Kupiki-Hotspot-Script #63: Change interface

Oscaretto, 2017-10-08:
Hey, I have installed the img on the raspberry pi 3, now i want use wlan0 as wan and wlan1 as lan, how to i change with the preinstalled image?

Thanks You :)

pihomeserver, 2017-10-08:
Hello
The script used is an old version. Changing interfaces after the installation is not the simplest path. You have to update : 
- /etc/network/interfaces
- /etc/chilli/defaults
- /etc/hostapd/hostapd.conf

Try to replace values in the files and restart all services/interfaces or a reboot

Tell me if that made the job

Oscaretto, 2017-10-08:
i changed in
/etc/chilli/defaults
/etc/hostapd/hostapd.conf

but in the /etc/network/interfaces I don't know what I must change

pihomeserver, 2017-10-09:
In the script there is the following code : 
```
if [ $COMMAND_RESULT -ne 0 ]; then
cat >> /etc/network/interfaces << EOT

auto $LAN_INTERFACE
allow-hotplug $LAN_INTERFACE
iface $LAN_INTERFACE inet static
    address $HOTSPOT_IP
    netmask 255.255.255.0
    network $HOTSPOT_NETWORK
    post-up echo 1 > /proc/sys/net/ipv4/ip_forward
EOT
    check_returned_code $?
fi
```
Set this configuration not to wlan0 but to wlan1
Also update wlan0 configuration according to your configuration

I think it would be much easier to perform that with the script on a fresh image instead of using the out-of-date one that you downloaded

pihomeserver, 2018-02-12:
As explained, better to configure before executing the script due to different impacts. Please reopen is needed

closed 2018-02-12 by pihomeserver