pihomeserver/Kupiki-Hotspot-Script #64: Oops, something went wrong with the configuration file

travuga77, 2017-10-10:
Good afternoon!

After 1.4.4 update pihotspot.sh script cannot find configuration.json file at "Updating Captive Portal" file step. As I understand, Kupiki-Hotspot-Portal is updating right now, so configuration file will appear later, isn't it?

pihomeserver, 2017-10-10:
My fault my fault ! I forgot yesterday to push the configuration.json file in the portal GitHub !
You can download it manually or restart the script

fernas, 2017-10-10:
Now I'm getting an error on "Cloning daloradius project".

pihomeserver, 2017-10-10:
more details please !

fernas, 2017-10-10:
On Log:

fatal: unable to access 'https://github.com/lirantal/daloradius.git/': Could not resolve host: github.com.

When I start this script, my dhcp setting are not working anymore and my raspberry pi lost its internet connection.

pihomeserver, 2017-10-11:
Strange as the script is not updating any DNS information. But the /etc/network/interfaces file is updated to configure the WLAN interface only. Can you please share the content of this file ?

travuga77, 2017-10-11:
my problem is solved, thanks!

closed 2017-10-11 by travuga77