pihomeserver/Kupiki-Hotspot-Script #65: login/psw

ojon1st, 2017-10-11:
Bonjour,
je viens d'installer la dernière version du script et en me connectant à mon nouveau hotspot, je suis redirigé  vers la page d'accueil de daloradius. Seulement les ids précisés à la fin du script à l'installation à savoir administrator/radius me renvoie une erreur. login or password incorrect. Que puis-je faire pour accéder au portail?
Merci d'avance.Cdlt

pihomeserver, 2017-10-11:
Bonjour
En vous connectant à votre hotspot vous devez avoir le portail captif (image de fond de la ville de Luxembourg de nuit). Ce n'est pas Daloradius. L'interface d'administration est accessible via l'url donnée à la fin de l'execution du script : http://<ip de votre raspberry>/

ojon1st, 2017-10-11:
Ok Merci de votre réponse cependant l'interface demande un identifiant et un mot de passe. Quels sont - ils?

pihomeserver, 2017-10-11:
Quelle interface ? 
Celle du hotspot : il faut les créer dans daloradius
Celle de daloradius : c'est ceux donné par le script (administrator / radius)

ojon1st, 2017-10-11:
Alors une fois que le script est installé, je me connecte à mon pi et je tape l'adresse ip de mon pi. Jusque là tout est ok. Je suis ensuite redirigé vers le portail exactement le meme que celui que vous avez décrit plus haut (image de fond de la ville de Luxembourg de nuit). Là je ne sais plus comment quels identifiants renseigner afin de pouvoir me connecter.

ojon1st, 2017-10-11:
Re. Désolé pour mon dernier post parce que vous aviez donné la réponse. Mille excuses. 
pouvez-vous confirmer la démarche après installation ou corriger svp:
1. se connecter à daloradius (http://192.168.10.1/daloradius) afin de créer les id d'accès au portail (KH Portal). Il faut etre connecter à pihotspot pour se faire.
2. se connecter au portail (KH Portal) et renseigner les id créés au 1.

Merci

pihomeserver, 2017-10-11:
Non pour des raisons de sécurité Daloradius n'est pas accessible par le hotspot. Uniquement depuis l'interface WAN (interface qui a internet)
Et ensuite vous utilisez les identifiants créés dans le hotspot

ojon1st, 2017-10-12:
Bonjour,
en suivant toutes vos instructions, j'ai une erreur 404 quand j'essaie de me connecter à la page Daloradius (http://ip/daloradius)

pihomeserver, 2017-10-12:
Il n'y a pas de daloradius à la fin de l'url de connexion

ojon1st, 2017-10-12:
ok comment puis-je alors me connecter à daloradius pour créer mon identifiant à renseigner sur le portail?

ojon1st, 2017-10-12:
j'essaie vraiment de trouver seul, je fais des tests et des recherches mais je ne trouve pas de solution. 
concrètement j'en suis à l'installation du script qui est ok.

quelles sont les étapes à suivre ensuite:
(si vous pouviez prendre 2 min pour les détailler ce serait parfait ).

merci d'avance

pihomeserver, 2017-10-12:
Il faut se connecter sur l'ip de votre raspberry 
Quelle est l'ip de votre raspberry ? Pas l'ip du hotspot ! C'est l'ip utilisée pour vous connecter et lancer le script
Une fois connue, dans un navigateur il suffit de se connecter en http avec cette ip.

pihomeserver, 2017-10-12:
Vous avez aussi l'information à la fin de la log du script (dans /var/log)

ojon1st, 2017-10-13:
Super, je vous remercie.

closed 2017-10-13 by pihomeserver