pihomeserver/Kupiki-Hotspot-Script #66: Unable to find chilli interface tun0

travuga77, 2017-10-11:
At the end of script it cannot find tun0:

```
::: Starting CoovaChilli service
::: Checking if interface tun0 has been created by CoovaChilli
::: Unable to find chilli interface tun0
```

`ifconfig -a` do not show any interface named tun0 too. But when I restart chilli with `sudo /etc/init.d/chilli restart`, it appears. What's wrong?

pihomeserver, 2017-10-11:
the tun0 interface is created by Chilli. If it's not found that means that chilli is not started.
Can you please confirm that after a reboot, chilli is up and running (so that means that you can find tun0 in the interfaces returned by ifconfig)

travuga77, 2017-10-11:
After reboot everything is ok with `ifconfig -a`.
Debian Stretch, RPi3. 
Yes, I've started script 4 or more times.

pihomeserver, 2017-10-11:
Ok so i guess there is an issue with chilli to get started after few execution of the script.
But at least if works ...

closed 2017-10-11 by pihomeserver

travuga77, 2017-10-11:
my mistake - I did not understand that HOTSPOT_IP is LAN ip, not WAN. After correcting it script worked ok.