pihomeserver/Kupiki-Hotspot-Script #67: Embed username and password to QRcode

bboy1983, 2017-10-29:
Hi, Thanks for the Creator of this Project.
i really appreciate your work.

anyway do you have any idea on how to embed of auto log-in using QRCODE

`qr = pyqrcode.create('http://10.1.0.1:4990/logon?username=%s&password=%s'% (username, password))`  

i have tried but, it not working, look like it also generate code another code for you to be log-in successfully..

pihomeserver, 2017-10-29:
Hello
That could be interesting to implement but not easy as i would like i think. In the current portal, the submit of login/password are first checked then push to a javascript function _connect_ in order to ask coova to validate credentials.
One way could be to create a dedicated page waiting for 2 parameters (login/password) and pushing them to coova after initializing the CoovaLibrary
Depends of your needs but doing this seems to me a little bit unsafe as someone could do a brute force hack using that page to guess credentials in particular if you don't want users to validate terms of use

bboy1983, 2017-10-31:
Or maybe we can do it very simple.
The easiest way i think, "auto fill" once user Scan the QR code, the it will redirect into log-in form with username and password already filled the form, then hit submit.
I have't tried a couple of times, yes it fill the form with correct user & pass but when i hit submit, it just return to the log-in page.

pihomeserver, 2017-10-31:
Please update your portal with the latest version (0.6). You can replace the portal.js with the new version.
I changed few elements and my first tests seem to be ok

pihomeserver, 2017-10-31:
If you try to input same credentials manually then you don't get the problem ?
Could be the reading of the inputs or that the connection to Coova has not been established yet.