pihomeserver/Kupiki-Hotspot-Script #70: No Internet Connection

bboy1983, 2017-11-10:
Good Day

I'm connected successfully with the Hotspot but there is no internet connection. Can't browse or anything.

pihomeserver, 2017-11-10:
Hello. Without more details than "it does not work" it's difficult to help.
What script version ? What os ? Did you get errors with the script when. installing ? Is coova running ? Do you see tun0 network interface ?

jkanasu, 2017-12-15:
Same behaviour with my setup as well.
I downloaded (git clone) the script on Dec 14, 2017(i.e. v1.4.9)
Ran the script on  on raspbian stretch 2017-11-29 release and kernel version 4.9
The script had installation problems, 

- first it would not install mariadb-server, so i manually installed (apt-get) and reran the script
- Next problem was, it was not able to bring up the wlan0, the "RTNETLINK answers: File exists " error was thrown, so i "ip addr flush dev wlan0" and eventually was able to ifup/ifdown the wlan0
Reran the script and it went fine.
I check that the radius server was running, coova running, was able to configure new users in the system.
Now the hotspot is visible for other devices, we can connect to the pi hotspot, it allows us to login, and it says succesfully connected.
But then further there is no internet connectivity.
I have the tun0 running, see details below.
NOTE: My internet connection is through the ethernet using PPPoE
Help appreciated

root@pavanpihotspot:~/github/Kupiki-Hotspot-Script# cat /etc/network/interfaces
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto dsl-provider
iface dsl-provider inet ppp
pre-up /bin/ip link set eth0 up # line maintained by pppoeconf
provider dsl-provider

auto eth0
iface eth0 inet manual

auto wlan0
allow-hotplug wlan0
iface wlan0 inet static
    address 192.168.10.1
    netmask 255.255.255.0
    network 192.168.10.0
    post-up echo 1 > /proc/sys/net/ipv4/ip_forward

root@pavanpihotspot:~ $ ifconfig 
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fea7:cc1e  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:a7:cc:1e  txqueuelen 1000  (Ethernet)
        RX packets 810  bytes 819764 (800.5 KiB)
        RX errors 0  dropped 73  overruns 0  frame 0
        TX packets 604  bytes 66244 (64.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 73  bytes 15534 (15.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 73  bytes 15534 (15.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ppp0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1492
        inet 172.31.49.242  netmask 255.255.255.255  destination 203.192.247.198
        ppp  txqueuelen 3  (Point-to-Point Protocol)
        RX packets 714  bytes 807666 (788.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 575  bytes 45054 (43.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

tun0: flags=81<UP,POINTOPOINT,RUNNING>  mtu 1500
        inet 192.168.10.1  netmask 255.255.255.0  destination 192.168.10.1
        inet6 fe80::3cee:2f7b:b95e:fd5d  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 100  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 23  bytes 1452 (1.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4419<UP,BROADCAST,RUNNING,PROMISC,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fef2:994b  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:f2:99:4b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 23  bytes 3214 (3.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

pihomeserver, 2017-12-15:
The script is used to be installed on the lite version of Raspbian. The "standard" one has differences.

About internet once logged in via SSH can you pping any internet website ?
If yes, can you please share the /etc/coova/up.sh script content ?

pihomeserver, 2018-02-12:
As said, please use Raspbian Lite to run the script. If issue persists please reopen

closed 2018-02-12 by pihomeserver