pihomeserver/Kupiki-Hotspot-Script #71: Error on script run: "Could not resolve host: github.com"

AdrianMrn, 2017-11-22:
Heya!

I'm installing the script on the latest version of Raspbian Lite and halfway through the installation, when the script is running `Cloning CoovaChilli project`, I get the error `fatal: unable to access 'https://github.com/coova/coova-chilli.git/': Could not resolve host: github.com`.

I saw some other people had this same issue, is there a fix? Or am I doing something wrong? :)

Thanks

pihomeserver, 2017-11-22:
Hello
You found the reason yourself. You can't use same NIC for hotspot and internet at the same time.
Also there is a generated iptable in the coova up.sh script that blocks all traffic on the api network for security reason.

Try to connect your ethernet NIC while installing the script and then update up.sh to remove the iptable rule. Keep in mind how you will be able to ssh your pi or update it ...

reigelgallarde, 2017-11-24:
Yes, I can confirm this is happening... it's because of the line where you will run a network restart and then without even finishing restart, the script calls for a git clone.

My workaround was to run the cloning first, then restart network, then continue with the rest...
In short, I move `Cloning CoovaChilli project` above `Restarting networking (via systemctl): networking.service.`

reigelgallarde, 2017-11-24:
It might be a good idea to create another function that clones or downloads the needed source files after installing the dependent packages.

Something like `download_source_files() { }`...
Then this will be called next to https://github.com/pihomeserver/Kupiki-Hotspot-Script/blob/master/pihotspot.sh#L342
````
install_dependent_packages PIHOTSPOT_DEPS[@]
download_source_files
````

pihomeserver, 2017-11-24:
@reigelgallarde thanks for the tip. I will work on that. Maybe this should be done at the end of the script to allow those who don't want to keep ethernet NIC to install the script.

AdrianMrn, 2017-11-24:
Thanks for the fast and incredibly helpful feedback, guys :)

I'll see if I can find some time to create a pull request

reigelgallarde, 2017-11-25:
@pihomeserver , what I meant was move all the lines that does all the cloning and downloading of source files... move it inside `download_source_files`

so at the very least, after these two functions, internet connection is not required... it will fixed this issue #71.

```
install_dependent_packages PIHOTSPOT_DEPS[@]
download_source_files
```

pihomeserver, 2017-11-26:
@AdrianMrn the script has been updated. You can give a try to it and tell us if it does what you were expecting

reigelgallarde, 2017-12-21:
I have tried the script and it works...

although, I'm not sure why but my pi can't get past this line
`execute_command "/etc/init.d/networking restart" true "Restarting network service to take IP forwarding into account"`

My pi would loss my connection because I'm on ssh..
So, when I'm on ssh, I just remove that line and do manual reboot after the script is done...
Just a note for everyone...

pihomeserver, 2018-02-12:
Last versions of the script perform well from start to end. Reopen if same issue still persists

closed 2018-02-12 by pihomeserver