pihomeserver/Kupiki-Hotspot-Script #72: Daloradius config into MySql

blakegrandmaster, 2017-11-27:
:::
::: You are root - OK
:::
::: Verifying free disk space (500 Mb) - OK
::: Preparing log file
::: Creating the log file
::: 
::: Log file created : /home/pi/Documents/hotspot/pihotspot.log
::: Use command 'tail -f /home/pi/Documents/hotspot/pihotspot.log' in a new console to get installation details
::: Updating APT config to force IPv4
::: Configuring localepurge
:::
::: Checking apt-get for upgraded packages.... done!
:::
::: Your system is up to date! Continuing with Pi-Hotspot installation...
::: Remove old configuration of Coova Chilli
::: Remove old configuration of haserl
::: Remove old configuration of hostapd
::: Checking for tun module
::: Insert tun module if existing (for Raspbian Jessie Lite)
::: Modprobe module (no check - useless if already loaded)
::: Checking for tun module
::: Checking if wlan0 interface already exists
::: Adding MariaDb password
::: Adding MariaDb password (confirmation)
::: Getting WAN IP of the Raspberry Pi (for daloradius access)
::: Adding Avahi dependencies
::: Updating the system hostname to wifiname
::: Updating /etc/hosts
:::
::: Checking apt-get for upgraded packages.... done!
:::
::: Your system is up to date! Continuing with Pi-Hotspot installation...
:::
::: Removing any previous sources of CoovaChilli project
::: Cloning CoovaChilli project
::: Cloning daloradius project
::: Cloning Pi Hotspot portal project
::: Starting MySql service
::: Update interface configuration (eth0)
::: Update interface configuration (wlan0)
::: Activating the WAN interface
::: Activating the LAN interface
::: Stopping freeradius service to update the configuration
::: Creating freeradius database
ERROR 1133 (28000) at line 1: Can't find any matching row in the user table
ERROR 1396 (HY000) at line 1: Operation DROP USER failed for 'radius'@'localhost'
::: Installing freeradius schema
::: Adding setup data
::: Updating freeradius configuration - Activate SQL support
::: Configuration of the Freeradius SQL driver
::: Change dialect of the Freeradius SQL driver to mysql
::: Configuration of the Freeradius SQL connection
::: Updating freeradius configuration - Activate SQL counters
::: Bug fix for SQL dialect once SQL Counters are activated
::: Updating inner-tunnel configuration (1)
::: Updating inner-tunnel configuration (2)
::: Updating freeradius default configuration (1)
::: Updating freeradius default configuration (2)
::: Checking freeradius configuration
::: Activating IP forwarding
::: Restarting network service to take IP forwarding into account
::: Building CoovaChilli package
::: Installing CoovaChilli package
::: Configuring CoovaChilli up action
::: Block access from LAN to WAN except portal
::: Activating CoovaChilli
::: Backup of default configuration file
::: Configuring CoovaChilli WAN interface
::: Configuring CoovaChilli LAN interface
::: Configuring CoovaChilli hotspot network
::: Configuring CoovaChilli hotspot IP
::: Configuring CoovaChilli authorized network
::: Removing CoovaChilli secret key
::: Updating UAMFORMAT
::: Updating UAMHOMEPAGE
::: Configuring CoovaChilli hotspot SSID
::: Add CoA support
::: Add firewall allowed port
::: Activating CoovaChilli on boot
::: Creating configuration file for hostapd
::: Configuring hostapd
::: Installing Daloradius in Nginx folder
::: Loading daloradius configuration into MySql
./pihotspot.sh: line 580: /usr/share/nginx/html/daloradius/contrib/db/fr2-mysql-daloradius-and-freeradius.sql: No such file or directory
::: 
::: Something went wrong with the last command. Please check the log file
::: 

Also

::: Configuring CoovaChilli hotspot SSID
::: Add CoA support
::: Add firewall allowed port
::: Activating CoovaChilli on boot
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
::: Creating configuration file for hostapd
::: Configuring hostapd
::: Installing Daloradius in Nginx folder
::: Loading daloradius configuration into MySql
::: 
::: Something went wrong with the last command. Please check the log file
:::

blakegrandmaster, 2017-11-27:
After running the script again

-I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT rtmon.lo -MD -MP -MF .deps/rtmon.Tpo -c rtmon.c  -fPIC -DPIC -o .libs/rtmon.o
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT rtmon.lo -MD -MP -MF .deps/rtmon.Tpo -c rtmon.c -o rtmon.o >/dev/null 2>&1
depbase=`echo cmdsock.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CC   --mode=compile arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json    -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF='"/etc/chilli.conf"' -DDEFPIDFILE='"/var/run/chilli.pid"' -DDEFSTATEDIR='"/var/run"' -DSBINDIR='"/usr/sbin"' -I../json  -DDEFLIBDIR='"/usr/lib/coova-chilli"'  -MT cmdsock.lo -MD -MP -MF $depbase.Tpo -c -o cmdsock.lo cmdsock.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT cmdsock.lo -MD -MP -MF .deps/cmdsock.Tpo -c cmdsock.c  -fPIC -DPIC -o .libs/cmdsock.o
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT cmdsock.lo -MD -MP -MF .deps/cmdsock.Tpo -c cmdsock.c -o cmdsock.o >/dev/null 2>&1
depbase=`echo chilli_module.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CC   --mode=compile arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json    -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF='"/etc/chilli.conf"' -DDEFPIDFILE='"/var/run/chilli.pid"' -DDEFSTATEDIR='"/var/run"' -DSBINDIR='"/usr/sbin"' -I../json  -DDEFLIBDIR='"/usr/lib/coova-chilli"'  -MT chilli_module.lo -MD -MP -MF $depbase.Tpo -c -o chilli_module.lo chilli_module.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT chilli_module.lo -MD -MP -MF .deps/chilli_module.Tpo -c chilli_module.c  -fPIC -DPIC -o .libs/chilli_module.o
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT chilli_module.lo -MD -MP -MF .deps/chilli_module.Tpo -c chilli_module.c -o chilli_module.o >/dev/null 2>&1
depbase=`echo ../extern/strlcpy.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
/bin/bash ../libtool  --tag=CC   --mode=compile arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json    -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF='"/etc/chilli.conf"' -DDEFPIDFILE='"/var/run/chilli.pid"' -DDEFSTATEDIR='"/var/run"' -DSBINDIR='"/usr/sbin"' -I../json  -DDEFLIBDIR='"/usr/lib/coova-chilli"'  -MT ../extern/strlcpy.lo -MD -MP -MF $depbase.Tpo -c -o ../extern/strlcpy.lo ../extern/strlcpy.c &&\
mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT ../extern/strlcpy.lo -MD -MP -MF ../extern/.deps/strlcpy.Tpo -c ../extern/strlcpy.c  -fPIC -DPIC -o ../extern/.libs/strlcpy.o
libtool: compile:  arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../json -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -I../json -DDEFLIBDIR=\"/usr/lib/coova-chilli\" -MT ../extern/strlcpy.lo -MD -MP -MF ../extern/.deps/strlcpy.Tpo -c ../extern/strlcpy.c -o ../extern/strlcpy.o >/dev/null 2>&1
/bin/bash ../libtool  --tag=CC   --mode=link arm-linux-gnueabihf-gcc -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF='"/etc/chilli.conf"' -DDEFPIDFILE='"/var/run/chilli [... cut ...]
::: Configuring CoovaChilli authorized network
::: Removing CoovaChilli secret key
::: Updating UAMFORMAT
::: Updating UAMHOMEPAGE
::: Configuring CoovaChilli hotspot SSID
::: Add CoA support
::: Add firewall allowed port
::: Activating CoovaChilli on boot
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults
::: Creating configuration file for hostapd
::: Configuring hostapd
::: Installing Daloradius in Nginx folder
::: 
::: Something went wrong with the last command. Please check the log file
:::

pihomeserver, 2017-11-27:
Arf. Something is wrong with current script version. I will restart tests.

pihomeserver, 2017-11-28:
Please download latest release 1.4.9 and restart the script

closed 2017-11-28 by pihomeserver