pihomeserver/Kupiki-Hotspot-Script #74: Cant connect to daloradius through IP given

AntPerez69367, 2017-12-03:
The script setup without any errors and the wifi hotspot works however I can't connect to the user administration through the ip given at the end of installation. http://<IP> or http://<IP>/daloradius. I also tried http://pihotspot.local but that also does not work. All attempts to connect are returning an unable to connect/ cant establish connection. Any ideas on where to start troubleshooting this?

pihomeserver, 2017-12-03:
Please connect to your pi with ssh and check that nginx is up and running and its logs where you should have some information about the error

AntPerez69367, 2017-12-04:
Thanks for pointing me in the right direction. I checked the service and it is running fine, there were no errors in the logs
```
nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2017-12-03 11:47:29 EST; 35s ago
     Docs: man:nginx(8)
  Process: 1076 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS
  Process: 615 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0
 Main PID: 1078 (nginx)
   CGroup: /system.slice/nginx.service
           ├─1078 nginx: master process /usr/sbin/nginx -g daemon on; master_process on;
           ├─1079 nginx: worker process
           ├─1080 nginx: worker process
           ├─1081 nginx: worker process
           └─1082 nginx: worker process

Dec 03 11:47:24 pihotspot systemd[1]: Starting A high performance web server and a reverse proxy server..
Dec 03 11:47:29 pihotspot systemd[1]: Started A high performance web server and a reverse proxy server.
```
I then checked the nginx folder and I saw that default was present in sites-available but was there was no default in sites-enabled. I did `ln -sfT /etc/nginx/sites-available/default /etc/nginx/sites-enabled/default` and now I am able to connect like I should be.

reigelgallarde, 2017-12-05:
use the ip you've used for connecting to shh.

Example, if you have ssh using `192.168.1.20` then use `192.168.1.20/daloradius`

closed 2017-12-20 by AntPerez69367