pihomeserver/Kupiki-Hotspot-Script #75: Captive portal without ethernet

Tywors, 2017-12-30:
Hello!
Why don't show the captive portal if the raspberry don't have internet (ethernet cable)?

How I can resolve this, please?

pihomeserver, 2017-12-30:
Sorry i Don't understand what is your expected behavior. Can you please provide more details

Tywors, 2017-12-30:
Hi!
I need kupiki only for navegate in Lan websites, for wikis and docs. But if i disconnect the ethernet from the raspberry, i can connect to hotspot but don't show the captive website.

Sorry for my english, 

Cool project! thanks

pihomeserver, 2017-12-31:
Ok so let me guess why it does not work. If you remove the ethernet link, then the interface is not set with an IP, etc. Then coova rules in up.sh with IP tables will not work at all.
Try to remove added rules in up.sh.

Tywors, 2017-12-31:
Thanks! I try tomorrow! Where is the up.sh?

pihomeserver, 2018-01-01:
You will have to remove the added rules at the end of the script
If there is no cable i'm pretty sure that the IP will not be set and then redirection rules will fail

Tywors, 2018-01-06:
Hi, i'm using your prepare .img
Would you know what number of lines they are?

I'm not able to make it work. Thanks!

pihomeserver, 2018-01-06:
The image is obsolete. Lines are the one added to /etc/coova/up.sh Should be the 8-10 lines at the end. I dont have the hotspot available for now
-------- Message d'origine --------De : Victor Maldonado <notifications@github.com> Date : 06/01/2018  17:32  (GMT+01:00) À : pihomeserver/Kupiki-Hotspot-Script <Kupiki-Hotspot-Script@noreply.github.com> Cc : Pi Home Server <pihomeserver@gmail.com>, Comment <comment@noreply.github.com> Objet : Re: [pihomeserver/Kupiki-Hotspot-Script] Captive portal without ethernet (#75) 
Hi, i'm using your prepare .img

Would you know what number of lines they are?
I'm not able to make it work. Thanks!

acujl, 2018-09-21:
Erased this lines
`ipt -I POSTROUTING -t nat -o $HS_WANIF -j MASQUERADE

LOCAL_IP=`ifconfig $HS_WANIF | grep -Eo "inet (addr:)?([0-9]\.){3}[0-9]" | gr$
LOCAL_LAN=`ip -o -f inet addr show | grep $HS_WANIF | awk '/scope global/ {prin$
ipt -I FORWARD 1 -i $TUNTAP -d $LOCAL_LAN -j DROP
ipt -I INPUT 1 -i $TUNTAP -d $LOCAL_IP -j DROP`

Hotspot on, captive portal dead

acujl, 2018-09-21:
Put them back, and now captive portal is still dead '-'

pihomeserver, 2018-09-22:
Dis you stop - start the service ? Reboot ? If it still does not work then you sure changed something else before

acujl, 2018-09-22:
Just reinstalled everything. working.

Now I'm trying to see if this work without Ethernet plugged in

acujl, 2018-09-23:
Any thoughts on that? @pihomeserver

acujl, 2018-10-02:
Found this: https://github.com/coova/coova-chilli/issues/212

Any suggestions? Still nothing when I unplug the wlan0

pihomeserver, 2018-10-02:
Sounds to help in the good direction with the capability to have a landing page in case Internet is not up. If you give a try i'm curious of the result to add it in the script

s13884, 2019-04-18:
@Tywors  @acujl  Are you able to configure the chilli without internet?

Tywors, 2019-04-18:
@s13884 yes, without internet yes. But i need connect (raspberry in my case) to a wifi mobile phone without internet. But work :)

s13884, 2019-04-19:
@Tywors So it requires Ethernet cable plugged in without internet? Or rpi wifi connected to AP?  
Also would you please help me to configure the same?

Tywors, 2019-04-19:
@s13884 sorry i don't have time to help.. but its simple i only connect rpi to AP without internet.

s13884, 2019-04-19:
Ok @Tywors, is it possible for you to share the chilli config?

doxendine, 2019-12-17:
Hi. I have a similar question. I'm trying to setup a captive portal with only a Raspberry Pi. No internet & no router. 
The purpose is (in a no wifi situation) to have a custom portal page to allow users to connect via wifi to the page & download different files on the Raspberry Pi.
Do you know if this is possible via Kupiki or other option?

Thanks for your help.