pihomeserver/Kupiki-Hotspot-Script #76: session time/session download not working

sherwyng, 2018-01-13:
Hi, thanks for this great script you have published. 

I have used your previous script from 2017 and that worked wonderfully. I noticed you have made a few changes/updates, so I attempted to use your current script.

The installation once again works wonderfully, however, I noticed that the daloradius platform  is unable to track  how much of data a user has transferred/the last user login in/ session time/session download/etc. You previous installation allowed for this.

Can you kindly advise

pihomeserver, 2018-01-13:
Hello
How do you control that ? Within Daloradius or with special counters ?

sherwyng, 2018-01-13:
Hi,  I was under the impression  that coovachilli tracks the users session  time and allows for session time/data  limits to be set.  These are stored in a database and daloradius is merely a portal to view this information.  As I previously  mentioned  your previous install file,  installed with this functionality.  This install file doesn't  seem to do this

pihomeserver, 2018-01-14:
Ok so i guess that coova counters are not defined any more. It's an SQL to load. Daloradius comes from its good counters for Freeradius 2.x but new system uses Freeradius 3.x. The SQL has to be updated if needed and i missed that point ...

I don't have the pi available so can't help now but you can try this command once you updated the SQL within Daloradius folder : 
```
mv /etc/freeradius/sql/mysql/counter.conf /etc/freeradius/sql/mysql/counter.conf.bak && cp /usr/share/nginx/html/daloradius/contrib/configs/freeradius-2.1.8/cfg1/raddb/sql/mysql/counter.conf /etc/freeradius/sql/mysql/counter.conf
```

sherwyng, 2018-01-14:
Hi thanks for your prompt response. After posting the question,  I browsed the internet and certain posts suggest  that, this is caused by Freeradius 3.x (the suggest using older versions). 

https://sourceforge.net/p/daloradius/discussion/684102/thread/56b11486/

Might I suggest that the ./pihotspot.sh include a  Freeradius 2.x (if possible).  Unfortunately with  Freeradius 3.0 a number of the file configurations have changed (I have tried the code above, as expected it was not successful). I will continue trying various things. When your pi is available, I would appreciate any feedback you can provide. If I achieve any success, I will report back.

Thanks again

pihomeserver, 2018-01-14:
I will check that. For now i prefer to keep Freeradius 3 to avoid issues with latest Debian Stretch
Thank you for your feedback if you succeed

sherwyng, 2018-01-20:
Hi again, have you had a chance to look into this. I have tried various things with no progress thus far. Or alternatively can you provide a link for the img file used

pihomeserver, 2018-01-21:
Sorry not yet. If you want the image with the old version the link is within the readme

reigelgallarde, 2018-01-22:
I have tried this too and can see the problem...
This issue should be addressed by daloradius if they ever planned to support freeradius 3.
Their config is applicable only from 2.1.8 up to 2.2 as I've checked it.

freeradius 3 has changed a lot in their config. I have tried migrating the config but have failed.
I've already reported this issue on their github: https://github.com/lirantal/daloradius/issues/46

Hope someone can find a solution soon as I also want to use rpi stretch and it has freeradius 3 on it's repository.

I'm at my end on migrating the config. I really have no clue on how to correctly do it. I'm now trying to edit my copy of pihotspot.sh and see if I can managed to install freeradius 2.x.x on rpi stretch. I know this will be a lot of work, but I'll give it a try.

Another possible solution I'm thinking is to stop using daloradius until they have a config for freeradius 3. I'm thinking of using [Dialup admin](https://wiki.freeradius.org/guide/Dialup-admin), I've read that this comes with freeradius 3. I will do this if ever I will fail with freeradius 2.x.x on rpi stretch.

pihomeserver, 2018-01-22:
Maybe we could go to dialup admin if you confirm that it supports freeradius 3. I work on a new web interface but if a good one already exists let use it

reigelgallarde, 2018-01-22:
I might scratch the idea of using  Dialup admin, I've researched further and it's now been removed on freeradius 3 installation. The reason that no one is maintaining it.

sherwyng, 2018-01-24:
Hi again, I tried the image in the readme file. It works when using an ethernet cable. When I attempt to use a 3G card (using a tutorial, I have previously used  https://nicovddussen.wordpress.com/2014/11/12/setting-up-your-raspberry-pi-to-work-with-a-3g-dongle/),  it doesn't work.  It works on stretch. Do you perhaps have another image file, I can try

pihomeserver, 2018-01-24:
FYI there are new counters to track the connection duration : http://networkradius.com/doc/3.0.10/raddb/mods-available/sqlcounter.html

About the image it's an old one with bugs and if you update/upgrade the bistro i'm pretty sure it will break the configuration
For now i do not have any up to date image. I first look at the counter issue

sherwyng, 2018-01-25:
Hi, I just wanted to say thanks for your prompt responses and efforts. I certainly do appreciate it

pihomeserver, 2018-01-25:
My first check found that radacct table is not updated. So counters will not work. Have to find why ...

pihomeserver, 2018-01-26:
Ok found a possible reason. With a fresh install the column acctupdatetime seems to not exist in radacct. And the driver in /etc/freeradius/3.0/mods-enabled/sql is still rum_sql_null.
The script has not generated errors and rerunning some commands work.

Try to change values manually and counters should work again. You will have to create yours for data volume manually until i add them

sherwyng, 2018-01-26:
Hi

Thanks again for your response, if I understand your correctly, I should look at these lines :

display_message "Updating freeradius configuration - Activate SQL support"
ln -sf /etc/freeradius/3.0/mods-available/sql /etc/freeradius/3.0/mods-enabled/sql
check_returned_code $?
chown -h freerad:freerad /etc/freeradius/3.0/mods-enabled/sql
check_returned_code $?

display_message "Configuration of the Freeradius SQL driver"
sed -i 's/"rlm_sql_null"$/"rlm_sql_mysql"/' /etc/freeradius/3.0/mods-enabled/sql
check_returned_code $?

display_message "Change dialect of the Freeradius SQL driver to mysql"
sed -i 's/"sqlite"$/"mysql"/' /etc/freeradius/3.0/mods-enabled/sql
check_returned_code $?

and attempt to do these manually if need be.

pihomeserver, 2018-01-27:
First have a look in radacct table if you have some values if some users already connect to your hotspot. If not :

Line 31 : you should have 
```
driver = "rlm_sql_mysql"
```

Line 87 : you should have with some differences if you changed default values on install 
```
dialect = "mysql"
server = "localhost"
port = 3306
login = "radius"
password = "radpass"
```

Then check that table radacct has the column acctupdatetime

Next, stop freeradius service and start freeradius -X
Then connect with a user and check that radacct is well populate. In the terminal running freeradius if you have red lines please share them

pihomeserver, 2018-01-27:
Ok found that daloradius is dropping radacct table to recreate it. Unfortunately it's based on the FR2 schema that was not using acctupdatetime so that's why new queries fail.
Will correct that soon. Then will see to add custom counters

sherwyng, 2018-01-27:
Hi again,  I followed what you said. And in my case the  acctupdatetime coloum was missing. So I added it, together with a acctinterval (as this was also reported missing). After doing this it appears to work. Thanks again for your help and the incredibly good work you doing.

pihomeserver, 2018-01-28:
Update committed.
Next will contain counters (data and time volume)

closed 2018-01-28 by pihomeserver

pihomeserver, 2018-01-29:
You can use CoovaChilli counters. Tested and validated (at least ChilliSpot-Max-Output-Octets)