pihomeserver/Kupiki-Hotspot-Script #77: Question: Download speed time allocation.

reigelgallarde, 2018-01-25:
I have searched many days but can't find the answer.

Is there a way have different download/upload speed based on time?

What we have now is we can make a user that has time expiry..
After time allocated being used up, the device will now have no internet and will need to login again.

What I'm looking for is that a user will have a time with max download speed.
So when the time is used up, I don't want the user be disconnected, but just lower the download speed.

Is this possible?

pihomeserver, 2018-01-25:
Even i don't have the answer, that could be cool to have an option to do that !

reigelgallarde, 2018-01-30:
a week passed.

I can say this is not a ready made option.

Possible solution I could think of:
1. Create custom sqlcounter. But this I'm not sure.
2. Create a daemon script that runs chilli_query authorize. Automatic login for not logged in connected users and give them minimal bandwidth. Problem is, we have to give the users an idea where to login.

vanrion, 2018-01-30:
That's possible, use the database and some simple script to change the
Bandwidth limit. you should have 2 pre-config bandwidth in daloradius,
change it in database using python script.

vanrion, 2018-01-30:
everything in daloradius is stored in database.

pihomeserver, 2018-01-30:
Also there are counters from coova that could reduce the bandwidth so seems to be possible in theory

It seems that counters have only one parameter assigned so not sure how to achieve the check for a limited time then limitied bandwidth

But like @vanrion said, all is SQL :-)

vanrion, 2018-01-30:
changing this with the second pre-configured bandwidth with limited speed,
this might work..
ex: 1st config "Bandwidth_limit" (max speed"
       2nd config "Bandwidth_limit_2" (with limited speed)

reigelgallarde, 2018-01-31:
@vanrion when you say 

> script to change the Bandwidth limit
do you mean update the database value for the limit?

Because if that's the case, chilli will set everything on login and whatever you change on database will not affect until the user logs in again...

pihomeserver, 2018-01-31:
Maybe one possible way is to use unlang in configuration file to update bandwith limit attribute. Still looking what can it's possible to achieve with it

About daloradius, no time for now to continue to work on the admin (https://github.com/Kupiki). I want to finish first the traffic logger for mandatory laws (at least France).
Help is welcome if you know about NodeJS and React ;)