pihomeserver/Kupiki-Hotspot-Script #78: Suggestion: Use domain name instead of IP for the portal.

reigelgallarde, 2018-01-30:
Would it be great to see `pihotspot.wifi/portal` or just `pihotspot.wifi` instead of the IP 192.168.10.1 when the users visit the portal.

It would be just like `pihotspot.local`.
I'm just not sure how to do it.

pihomeserver, 2018-01-30:
Two different points
- the .local is possible due to Bonjour protocol integrated in most OSes. It's for the administration part only. Adding a dedicated domain name requires DNS to allow other devices to resolve name to IP or to hardcode the name/ip in the host file of each device or to add that information to your router.

- Another point should be that Coova should be able to update the DNS to add the IP/name in his tables

For information/reminder, hardcoded url can help : logout, login and status. As soon as you input that as URL you will be redirected to the portal

reigelgallarde, 2019-01-09:
Bumping this, as I have found a possible solution...

https://blog.heckel.xyz/2013/07/18/how-to-dns-spoofing-with-a-simple-dns-server-using-dnsmasq/

the problem I'm having now is that it works if chilli is not running.

pihomeserver, 2019-01-09:
Can you please describe a little bit more ? The option uamanydns restricts by default to no other DNS than the 2 specified by dns1 and dns2 options. The current configuration blocks that too.

reigelgallarde, 2019-01-09:
the goal I'm for is to add `192.168.10.1   pihotspot.wifi` in the host file of dnsmasq.. it should then point pihotspot.wifi to 192.168.10.1 when browsing.. as IPs are hard to type for some people.. they prefer custom domain.. this works, but not when chilli is running.. I will test further..

pihomeserver, 2019-01-09:
If it does not work try to set dns1 and dns2 to the WAN interface IP to force chilli to forward requests to dnsmasq.