pihomeserver/Kupiki-Hotspot-Script #80: After a users data limit is reached, a user can log out and log in again and continue using data

sherwyng, 2018-02-09:
Hi, thanks once again for the good work and previous assistance you have provided.
I have encountered another problem I am trying to solve.  If you can assist with this it will be appreciated.
A user is assigned a specific data limit  (lets say 5 MB) with a reply attribute ChilliSpot-Max-total-Octets.
When the user reaches this data limit (5 MB)  he or she, internet acess is disconnected and the user is returned to the login screen, which is perfect. 
I expected the user would not now be allowed to logon as their data limit  (5 MB) has been reached.
However, I noticed  the user is allowed to login and continue using the data, until the data limit (10 MB) is reached once again.
So a user can continue login on and continue using data.
Any assistance you can provide with this matter is certainly appreciated.

pihomeserver, 2018-02-10:
Can you please check the sql request used by the counter ? Maybe that's the source

reigelgallarde, 2018-02-27:
can you please check what's inside your folder `/etc/freeradius/3.0/mods-config/sql/counter/<your sql choice>`

desevensa, 2018-06-27:
Try do that:
insert a line with this text 

`noresetcounter`

in the file 

> /etc/freeradius/3.0/sites-available/default

inside _authorize_ section. (Aprox line number 380).

After that reload radius or reboot the RPi.

If it works tell us, it can be inserted into the main script

pihomeserver, 2018-11-07:
@sherwyng any feedback to share on that ?

droecoleman, 2019-04-19:
@pihomeserver i will be testing this, i also encounter this problem now. I wonder if the solution has been reached,.

droecoleman, 2019-04-22:
@pihomeserver inserting a line noresetcounter inside authorize section as mentioned by @desevensa will solve the problem.

pihomeserver, 2019-04-23:
@droecoleman Did you change the counter ? The current configuration has only a max session time limit. Nothing related to data volume.
```
sqlcounter noresetcounter {
    sql_module_instance = sql
    dialect = mysql

    counter_name = Max-All-Session-Time
    check_name = Max-All-Session
    key = User-Name
    reset = never

    $INCLUDE ${modconfdir}/sql/counter/${dialect}/${.:instance}.conf
}
```

droecoleman, 2019-04-26:
my apology, the noresetcounter will solve the max-all-session. that has same occurrence as the problem posted. anyhow, if we can add chillispot_max_byes_noreset under authorize might solve the problem.

sqlcounter chillispot_max_bytes_noreset {
            counter-name = ChilliSpot-Max-Total-Octets
            check-name = Max-Octets
            reply-name = ChilliSpot-Max-Total-Octets
            reply-message = "You have reached your bandwidth limit"
            sqlmod-inst = sql
            key = User-Name
            reset = never
            query = "SELECT SUM(acctinputoctets) + SUM(acctoutputoctets) FROM radacct WHERE username='%{%k}'"

}

pihomeserver, 2019-04-26:
I see that all counters are missing. I will check that and update the installation script

pihomeserver, 2019-04-26:
Instead of noresetcounter you can add all followinf lines
```
dailycounter
weeklycounter
monthlycounter
quaterlycounter
yearlycounter
noresetcounter
expire_on_login
counterChilliSpotMaxTotalOctetsDaily
counterChilliSpotMaxTotalOctetsWeekly
counterChilliSpotMaxTotalOctetsMonthly
counterChilliSpotMaxTotalOctetsQuarterly
counterChilliSpotMaxTotalOctetsYearly
counterChilliSpotMaxInputOctetsDaily
counterChilliSpotMaxInputOctetsWeekly
counterChilliSpotMaxInputOctetsMonthly
counterChilliSpotMaxInputOctetsQuarterly
counterChilliSpotMaxInputOctetsYearly
counterChilliSpotMaxOutputOctetsDaily
counterChilliSpotMaxOutputOctetsWeekly
counterChilliSpotMaxOutputOctetsMonthly
counterChilliSpotMaxOutputOctetsQuarterly
counterChilliSpotMaxOutputOctetsYearly
```