pihomeserver/Kupiki-Hotspot-Script #87: login user management - JSON request timed out

MaAr389, 2018-03-02:
Hello, when I log in I get the following error message.
Any idea why that is?
[image]

pihomeserver, 2018-03-02:
Hello
The IP of the Coova server is not correct. What did you changed within the installation script ? Which parameters ?

MaAr389, 2018-03-02:
Hi, 
# IP of the hotspot
HOTSPOT_IP="192.168.10.212"
# Network where the hotspot is located
HOTSPOT_NETWORK="192.168.10.0"

The raspberry get a dhcp address from the linksys. 
At the linksys i define a dhcp reservation. So at the reboot the pi get the same ip.

pihomeserver, 2018-03-02:
The issue comes from the IP range. It's the same for eth0 and wlan0 so there is an issue on that.
You have to configure a different ranger wlan0. 
For example in your case 
```
HOTSPOT_IP=192.168.11.1
HOTSPOT_NETWORK=192.168.11.0
```
I guess i should find a way to check that by script to make a control when the script is executed

reigelgallarde, 2018-03-05:
@MaAr389, can you please paste here the content of your `/etc/network/interfaces`.
or enter command `cat /etc/network/interfaces`.

MaAr389, 2018-03-05:
@reigelgallarde sure.

`root@pihotspot:/home/pi# cat /etc/network/interfaces
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp

auto wlan0
allow-hotplug wlan0
iface wlan0 inet static
    address 192.168.10.212
    netmask 255.255.255.0
    network 192.168.10.0
    post-up echo 1 > /proc/sys/net/ipv4/ip_forward
`

reigelgallarde, 2018-03-06:
seems right to me...

@MaAr389 

can you check the output of

`service chilli status`

reigelgallarde, 2018-03-06:
I'm also concerned that your wlan0 has no IP... But the IP that I expected for it, is on eth0.

pihomeserver, 2018-03-06:
That's correct. The IP is set on the tun0 once Coova is started. As i said from the first post log the DHCP is providing an IP address that is in the same network range than the configurer IP for the hotspot. FMHO that's the reason why it does not work

reigelgallarde, 2018-03-07:
ahh yes.. The IP is on tun0. So the real problem is his `HOTSPOT_IP="192.168.10.212"`
This should be noted that in `192.168.xxx.yyy` IP format, `xxx` of eth0 and hotspot ip should not be the same.

desevensa, 2018-03-08:
I've got the same issue (even with default values). Looking the config files i realized that communication between _Radius_ and _Chilli_ were lost because in some part of the install script (of chilli) the secret key changed from **testing123** (in case of default values) to **aRandomSecureKey**. I fixed it restoring the "testing123" key in files **hs.conf** and **default** in _/etc/chilli._ After a reboot it worked great.
I don't (but I wish) know the total of files where testing123 is configured. In any case i guess it's simple to fix that in the script.
I hope this info be helpful and forgive my poor english.

Fakhreddine-Ghalleb, 2018-06-02:
To solve this issue just run a simple check on 3 service:

- service chilli status
- service freeradius status
- service mariadb status

My problem was that freeradius have not successfully started when i checked i got this:

`Jun 02 05:06:06 BeegoWIFI systemd[1]: Failed to start Start of freeradius after mysql.`

Then i stoped the service chilli and i run freeradius in Debug mod with **freeradius -X** and then i found this : 

`Failed binding to coa address * port 3799 bound to server coa: Address already in use 
/etc/freeradius/3.0/sites-enabled/coa[12]: Error binding to port for 0.0.0.0 port 3799`

So i changed the COA port to default **1700**

I hope that this will help someone.

CBeyssier, 2018-06-03:
I'm going to try the desevensa modification...
Thank you..

closed 2018-06-06 by pihomeserver

pihomeserver, 2018-09-28:
Hello
Look at the [reply](https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/87#issuecomment-394059003) from @Fakhreddine-Ghalleb : what about the chilli service which is the one that should respond ?

EasyVoiceTel, 2018-09-28:
Well the Chilly is very spicy xD

● chilli.service - LSB: Start CoovaChilli daemon at boot time
   Loaded: loaded (/etc/init.d/chilli; generated; vendor preset: enabled)
   Active: active (running) since Thu 2018-09-27 21:03:25 CEST; 21h ago
     Docs: man:systemd-sysv-generator(8)
  Process: 569 ExecStart=/etc/init.d/chilli start (code=exited, status=0/SUCCESS)
   CGroup: /system.slice/chilli.service
           └─691 /usr/sbin/chilli -c /etc/chilli.conf

Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[569]: chilli.
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[691]: PID 691 saving options to /var/run/chilli.691.cfg.bin
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[691]: PID 691 loading binary options file /var/run/chilli.691.cfg.bin
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[691]: Loading modules
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[691]: CoovaChilli 1.4. Copyright 2002-2005 Mondru AB. Licensed under GPL. Copyright 2006-20
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot chilli[691]: TX queue length set to 100
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot coova-chilli[693]: PID 693 loading binary options file /var/run/chilli.691.cfg.bin
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot coova-chilli[693]: Loading modules
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot coova-chilli[693]: USER root(0/0), GROUP root(0/0) CHILLI[UID 112, GID 117]
Sep 27 21:03:25 EasyVoice-Telecom-Hotspot coova-chilli[693]: Running /etc/chilli/up.sh (0/0)

EasyVoiceTel, 2018-09-29:
> I remade a test yesterday and i got no issue. Open a new issue and provide configuration parameters in particular ip adresses

Hi,

I Will but I didn’t made any changes to the install config, except the host name. 

Thank you! Merci! 😉

	
S.A.C. Abdel Nasser
Algemeen Directeur
T: +31(0)85 - 56 00 019
E: Shaady@EasyVoiceTelecom.nl | W: EasyVoiceTelecom.nl
			
De inhoud van dit bericht is alleen bestemd voor de geadresseerde en kan vertrouwelijke of persoonlijke informatie bevatten. Als u dit bericht onbedoeld heeft ontvangen verzoeken wij u het te vernietigen en de klantenservice, of de medewerker van EasyVoice Telecom direct te informeren. Het is niet toegestaan om een bericht dat niet voor u bestemd is te vermenigvuldigen dan wel te verspreiden. Aan dit bericht inclusief de bijlagen kunnen geen rechten ontleend worden, tenzij schriftelijk anders wordt overeengekomen. EasyVoice Telecom, aanvaardt geen enkele aansprakelijkheid voor schade en/of kosten die voortvloeien uit onvolledige en/of foutieve informatie in e-mailberichten.

pinoyoragon, 2018-10-03:
I also experience the same problem. I reinstalled everything, using "testing123" but nothing works. I also did the suggestions here but to no avail. Services (chilli, mariadb, freeradius) are enabled and running. Please help.

pihomeserver, 2018-10-03:
what is the output of ifconfig -a ? What is the content of the JSON file of the portal located in /usr/share/nginx/portal/js/configuration ?

EasyVoiceTel, 2018-10-03:
> I also experience the same problem. I reinstalled everything, using "testing123" but nothing works. I also did the suggestions here but to no avail. Services (chilli, mariadb, freeradius) are enabled and running. Please help.
> 
> #!/bin/bash
> 
> # PLEASE EDIT NEXT LINES TO DEFINE YOUR OWN CONFIGURATION
> # Name of the log file
> LOGNAME="kupiki_hotspot.log"
> 
> # Path where the logfile will be stored
> # be sure to add a / at the end of the path
> LOGPATH="/var/log/"
> 
> # Password for user root (MySql/MariaDB not system)
> MYSQL_PASSWORD="pihotspot"
> 
> # Name of the hotspot that will be visible for users/customers
> HOTSPOT_NAME="GaboWifi"
> 
> # IP of the hotspot
> HOTSPOT_IP="10.10.88.1"
> 
> # Wi-fi code country. Use above link to find yours
> # https://www.cisco.com/c/en/us/td/docs/wireless/wcs/3-2/configuration/guide/wcscfg32/wcscod.html
> WIFI_COUNTRY_CODE="PH"
> 
> # Use HTTPS to connect to web portal
> # Set value to Y or N
> HOTSPOT_HTTPS="Y"
> 
> # Network where the hotspot is located
> HOTSPOT_NETWORK="10.10.88.0"
> 
> # Secret word for FreeRadius
> #FREERADIUS_SECRETKEY=`cat /dev/urandom | tr -dc 'a-zA-Z0-9' | fold -w 32 | head -n 1`
> FREERADIUS_SECRETKEY="testing123"
> 
> # WAN interface (the one with Internet - default 'eth0' or long name for Debian 9+)
> WAN_INTERFACE=`ip link show | grep '^[1-9]' | awk -F ':' '{print $2}' | awk '{$1=$1};1' | grep '^e'`
> 
> # LAN interface (the one for the hotspot)
> LAN_INTERFACE="wlan0"
> 
> # Wifi driver
> LAN_WIFI_DRIVER="nl80211"
> 
> # Install Haserl (required if you want to use the default Coova Portal)
> # Set value to Y or N
> HASERL_INSTALL="N"
> 
> # Password used for the generation of the certificate
> CERT_PASSWORD="pihotspot"
> 
> # Number of days to certify the certificate for (default 2 years)
> CERT_DAYS="730"
> 
> # Make Avahi optional
> # Set value to Y or N
> AVAHI_INSTALL="Y"
> 
> # Install Daloradius Portal (compatible with FR2 only in theory)
> # Set value to Y or N
> DALORADIUS_INSTALL="Y"
> 
> # Enable/Disable Bluetooth
> # Set value to Y or N
> BLUETOOTH_ENABLED="N"
> 
> # Enable/Disable fail2ban to protect server from unwanted access
> # Set value to Y or N
> FAIL2BAN_ENABLED="Y"
> 
> # Enable/Disable Netflow logs to log all traffic requests. Must be crossed checked with assigned IP in the radius tables
> # Set value to Y or N
> NETFLOW_ENABLED="Y"
> 
> # Define how long Netflow logs will be stored
> # Sets the max life time for files generated for Netflow monitoring. The supplied maxlife_time accepts values such as 31d, 240H 1.5d etc.
> # Accepted time scales are w (weeks) d (days) H (hours).
> # A value of 0 disables the max lifetime limit. If no scale is given, H (hours) are assumed.
> # By default data are stored 365 days (value set to 365d)
> NETFLOW_LOGS_DAYS="365d"
> 
> # Enable/Disable MAC authentication
> # Set value to Y or N
> MAC_AUTHENTICATION_ENABLED="N"
> 
> # Password for MAC authentication. Could/Should be changed within the web administration interface
> MAC_AUTHENTICATION_PASSWORD="123456"
> 
> # Install web frontend of Kupiki Hotspot
> # Set value to Y or N
> INSTALL_KUPIKI_ADMIN="Y"
> 
> # Install Cron job for the hotspot updater. Will be executed every sunday at 8am (system time)
> # Set value to Y or N
> ADD_CRON_UPDATER="Y"

Wanna skip install?
https://github.com/pihomeserver/Kupiki-Hotspot-Script/issues/139

desevensa, 2018-10-03:
> Wanna skip install?
> #139

Set _secret_ as testing123 is not an issue, it was fixed sometime before my comment. Should be another thing, sorry

EasyVoiceTel, 2018-10-08:
Did it work?!